Klipper3d/klipper #3663: SKR 1.4 Turbo - can't install Klipper / connect via USB

vellkan, 2020-12-21:
Hello!

I've got problem with SKR 1.4 Turbo (on Ender 5 Pro).
I've install clean version of OctoPi 0.17 and Octoprint 1.5.2.
I've done everything via Klipper installation inctruction until I need to find USB port.
After send ls /dev/serial/by-id/* I've got something like that:
/dev/serial/by-id/usb-marlinfw.org_Marlin_USB_Device_10004010AF4A1C885D68A692F50020C0-if00

I found another issue with this, but author flash klipper.bin from SD Card. I've try that - download klipper.bin from RPi, change name, clean SD Card, put file and insert it to SKR. Now I've only hear buzzer from screen and that's all - Marlin wont's start, and Octoprint won't see mainboard.
After several  I've done something, because after ls/dev/serial I've got new info:
/dev/serial/by-id/usb-Klipper_lpc1769_10400010881C4AAF92A6685DC02000F5-if00
But again - without port information.

When I try flash from RPi (with make flash) I see this:

`Failed to flash to /dev/serial/by-id/usb-marlinfw.org_Marlin_USB_Device_10004010AF4A1C885D68A692F50020C0-if00: Error running dfu-util

If flashing Klipper to a Smoothieboard for the first time it may be
necessary to manually place the board into "bootloader mode" - press
and hold the "Play button" and then press and release the "Reset
button".

When a Smoothieboard is in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=1d50:6015

Alternatively, one can flash a Smoothieboard via SD card - copy the
"out/klipper.bin" file to a file named "firmware.bin" on an SD card
and then restart the Smoothieboard with that SD card.

make: *** [src/lpc176x/Makefile:36: flash] Error 255`

I've try few SD Cards (but I can flash Marlin at any time now), two USB cables, nothing works :/

I cannot upload klipper.log cause it doesn't exist.

Similar issue: https://github.com/KevinOConnor/klipper/issues/2805

Sineos, 2020-12-21:
> I found another issue with this, but author flash klipper.bin from SD Card. I've try that - download klipper.bin from RPi, change name, clean SD Card, put file and insert it to SKR. Now I've only [...]

This is correct. 
Enter in klipper config file:
```
[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1769_10400010881C4AAF92A6685DC02000F5-if00
```

x-c-dog-x, 2020-12-21:
I am having the same issue of not being able get a valid USB connection when installing MainsailOS on a SKR v1.4  Turbo. Klipper will not connect via USB.

vellkan, 2020-12-21:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5726326/klippy.log)

OK, now I've attached file.

@Sineos - nope, I put that to config file but nothing happens, but still - I cannot flash firmware.

vellkan, 2020-12-21:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5726342/klippy.log)

I've done everything again and upload a fresh log file.

Sineos, 2020-12-21:
I'm running Klipper on a SKR 1.4 Turbo myself. I would advise to delete Klipper, do a fresh check out, then:

1. `make menuconfig`
  - Micro-controller Architecture (LPC176x (Smoothieboard))  
  - Processor model (lpc1769 (120 MHz))
  - [*] Target board uses Smoothieware bootloader
  - [*] Use USB for communication (instead of serial)
 2. make
 3. rename `klipper.bin` to `firmware.bin`
 4. copy `firmware.bin` to SD-card
 5. Put SD-card into the SKR and reset
 6. Wait a bit
 7. Reset again
 8. Check SD-card: `firmware.bin` is replaced by `firmware.cur`
 9. `ls /dev/serial/by-id/*` 
   - Should look like `/dev/serial/by-id/usb-Klipper_lpc1769_10400010881C4AAF92A6685DC02000F5-if00`
10. Put the `[mcu]` into the config

```
[mcu]
serial: /dev/serial/by-id/usb-Klipper_lpc1769_10400010881C4AAF92A6685DC02000F5-if00
```

Your log:
```
[mcu creality]
serial = /dev/serial/by-id/usb-marlinfw.org_Marlin_USB_Device_10004010AF4A1C885D68A692F50020C0-if00
```
This indicates that still Marlin is on the board. Follow above guide.

NBouquain, 2020-12-21:
Only older versions of the SKR 1.3 board support USB flashing (they where shipped with the vanilla Smoothieware bootlader), the newer version have a patched bootloader that has removed this functionality due to pin conflict. See details in #2156
My guess is that SKR 1.4 still does not support USB flashing and you need to follow the procedure detailed by @Sineos 

Alternatively, in order to be able to flash firmware through the USB, you may try to flash the patched bootloader developped by @Arksine: https://github.com/Arksine/LPC17xx-DFU-Bootloader/releases

**Caution: it is known to work on SK v1.3 board but I am not sure about v1.4 board**

vellkan, 2020-12-21:
I've try bootloader from this topic:
https://github.com/ardiehl/BTT_SKR_13_14_14T_SD-DFU-Bootloader
But it didn't work as well.
I will keep looking, but I know people for whom SKR 1.4 & Klipper works :/

Sineos, 2020-12-21:
Just my personal opinion: I would refrain from playing with the bootloader, if you do not manage to install vanilla klipper

vellkan, 2020-12-21:
OK, I think it's working now.
I've try Your method @Sineos again (because I've done it before), but I was wating till buzzer in screen just turn off (just like flashing Marlin). It's not gonna happen, so I was turn off printer after 15-20 seconds.
Then I found that value: /dev/serial/by-id/usb-Klipper_lpc1769_10400010881C4AAF92A6685DC02000F5-if00, save it to printer.cfg and everything was fine, Klipper connect to printer :)

My friend send me his settings, but few days ago was major changes so I need join Ender 5 configuration with SKR 1.4 & TMC2009 & BMG extruder.

I think it's good idea to add in manual that people shouldn't wait for screen turn off when flashing file and -port0 is not necessary information in that command.
Last question - I saw on YT that after flashing Klipper I can still have Marlin, but I could do that - it is correct?

JeffMedd, 2020-12-22:
Hi all, Was having the same issue myself after the update to v0.91 of Klipper. I have a SKR 1.4 Turbo connected to a Raspberry Pi 3 using the 3 wire serial connection not USB. I can confirm that following Sineos guide above (but leaving unchecked 
 [ ] Use USB for communication (instead of serial) and not doing steps 9 &10) worked for me. Basically think there were some messed up config files and it was the uninstall (sudo ~/klipper/scripts/klipper-uninstall.sh
rm -rf ~/klippy-env ~/klipper)  and re-install that cleared this. Just thought I would put this in case others who are using the 3 wire connection see your thread.

vellkan, 2020-12-25:
I've got last question.
When I upload Klipper to SKR should I have something on my LCD screen after reboot? 
Now I've got blank screen, nothing more. I can control printer from web browser, but on Youtube I saw that people still have Marlin with some extra options in menu.

Tomblarom, 2020-12-27:
@vellkan try to do everything step by step. Don't mess with additional hardware (stepper, display, modules), until klipper is up and running. Having too many things to troubleshoot keeps you stuck.

vellkan, 2020-12-27:
> @vellkan try to do everything step by step. Don't mess with additional hardware (stepper, display, modules), until klipper is up and running. Having too many things to troubleshoot keeps you stuck.

I've succesfully connect Klipper to my Ender 5 Pro :) But I've stuck with configuration, especially with extruder calibration after last changes in firmware.
And I've seen issue with cutting & soldering TMC to work in UART so I think I pass for now, until I'm gonna be sure to left Marlin at all for Klipper.
I've try KIAUH too, but after home Y axis and hit to frame, my stepstick was move with some noise, so I need to back to Marlin.
Maybe I will try this when I build VORON or I find good and working configuration file for this settings :)

ThomasW69, 2020-12-30:
If you compile klipper for SKR1.4 you will get a file called klipper.bin in the /home/pi/klipper/out directory. You have to copy this file on an SD card, rename it to firmware.bin and stick the sd card into your SKR. Whe it poers up the firmware will be flashed. After successful flashin the file on the SD card is renamed to firmware.cur.

mrx23dot, 2021-01-08:
Stupid question, is it possible to connect to SKR boards via an onboard UART, instead of virtual USB UART? 
USB is not known to be reliable! Thus not wanted in 3D printing.

vvchik, 2021-01-12:
@mrx23dot yes it is possible using three wires connected to onboard uart. you need to follow @JeffMedd instruction in that case.

mrx23dot, 2021-01-12:
@vvchik  Awesome, on SKR 1.3 UART is on AUX-1 TFT header, thus it's not colliding with LCD.

vvchik, 2021-01-15:
one to add if you seen
 ```[mcu creality]
serial = /dev/serial/by-id/usb-marlinfw.org_Marlin_USB_Device```
and then did all @Sineos recommendations. And now see nothing in /dev/serial try to reconnect usb cable into another raspberry port. Not just disconnect/connect.

KoviRobi, 2021-01-17:
I have had success flashing a 1.4 via USB using DFU, the bootloader isn't a vanilla Smoothieware one, the DFU button has been moved to be the button of the rotary encoder. I have put my experiences [here](https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/compare/master...KoviRobi:patch-1).

closed 2021-02-15 by github-actions[bot]