Klipper3d/klipper #3665: Internal error on command:"TEMPERATURE_WAIT"

Block137, 2020-12-22:
Firmware shutdown instantly when executing a valid TEMPERATURE_WAIT command (invalid command returns error message but not shutdown)

```
Internal error on command:"TEMPERATURE_WAIT"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 177, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 115, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/heaters.py", line 340, in cmd_TEMPERATURE_WAIT
    temp, target = sensor.get_temp(eventtime)
AttributeError: PrinterExtruder instance has no attribute 'get_temp'
Transition to shutdown state: Internal error on command:"TEMPERATURE_WAIT"
Dumping gcode input 8 blocks
```

[klippy.zip](https://github.com/KevinOConnor/klipper/files/5729646/klippy.zip)

KevinOConnor, 2020-12-22:
Thanks.  Should be fixed now (commit a637c2f1).

-Kevin

Block137, 2020-12-23:
Thanks, it's working now

closed 2020-12-23 by Block137