Klipper3d/klipper #3666: Heaters Stay on after "Move exceeds maximum extrusion"  with hotend melting the part.

bung69, 2020-12-22:
Hi,  Just had my printer stop due to a "Move exceeds maximum extrusion" error.  - my fault as I forgot to change the nozzle size in the config.  no problem there.  

However the printer just stopped without disabling heaters.  is that expected behaviour?  I assume / hope the heaters will eventually time out?

Thanks

Matt

bung69, 2020-12-23:
[KlipperLog- move exceeds maximum extrusion.txt](https://github.com/KevinOConnor/klipper/files/5733608/KlipperLog-.move.exceeds.maximum.extrusion.txt)

jman1o1, 2021-01-20:
Just experienced the exact same scenario. Seems like a major safety issue.

monoxide13, 2021-02-04:
> Just experienced the exact same scenario. Seems like a major safety issue.

Same

KevinOConnor, 2021-02-26:
Responding to an error like this is typically handled by OctoPrint.  However, I agree Klipper could do a better job now that many users are using moonraker and mainsail/fluidd.  Adding a better default behaviour is something we will look at adding.

-Kevin

dudeofawesome, 2022-01-02:
+1 from me too. Just had it destroy a build platform due to leaving the heater and bed on with the nozzle oozing PETG onto the bed. Definitely should fail-safe with the heaters turning off in this event.

vladbabii, 2022-03-08:
Could we have an option to run some gcode on errors that stop the print, with an error code as a parameter? This way we can handle any possible issues in the best way for that printer

Sineos, 2022-03-08:
> Could we have an option to run some gcode on errors that stop the print, with an error code as a parameter? This way we can handle any possible issues in the best way for that printer

I second such a flexible approach. There is one instance, where I really would like to have the (bed)heater staying turned on:
With one of my printers I'm using 
`SCREWS_TILT_CALCULATE MAX_DEVIATION=0.02`
which tends to error out if the heat bed is not fully heat-soaked (or if I had to force off the last print from the bed). Here it would be really annoying if the heater gets turned off.

hamster65, 2022-03-08:
It would be usefull to me to switch off a wifi plug when there is an error with the heaters or sensors. Should be possible with this idea.... Right now it switches off for any error, such as "movement out of range". Very annoying, because it cuts all power, including the raspberry.

bwittig, 2022-03-22:
The heaters also stay on after a TMC error (GSTAT 0...1) - A similar solution would be good for that too.
(Still chasing the error - probably a bad connection on the motor?)

bernhardberger, 2022-04-25:
> Responding to an error like this is typically handled by OctoPrint. However, I agree Klipper could do a better job now that many users are using moonraker and mainsail/fluidd. Adding a better default behaviour is something we will look at adding.
> 
> -Kevin

Is it possible to have klipper itself handle security features? I find it a bit "dangerous" that it's handled by the frontend.

pedrolamas, 2022-05-08:
The bare minimum I would expect is for something like this to call `CANCEL_PRINT` - or at least give me an option to make it so!

Best case scenario, I would propose a new `ERROR_HANDLER` macro would run and inside of it we would check what the error was and handle however we want to (most likely, call `CANCEL_PRINT`)

pedrolamas, 2022-06-03:
With #5491 now merged, we now have a way to mitigate the above issue, by adding the following to the `virtual_sdcard` section:

```
[virtual_sdcard]
on_error_gcode:
  CANCEL_PRINT
```

This will be enough to make sure that a "Move exceeds maximum extrusion" error cancels the print and turns all heaters off!

OzzieOrca, 2022-06-08:
I experienced this with SuperSlicer due to this Monster Gap Fill issue https://github.com/supermerill/SuperSlicer/issues/2214.

I'll try adding that on error hook, thanks :)

kjkent, 2023-03-08:
Just commenting here after reading https://github.com/Klipper3d/klipper/issues/5430#issuecomment-1101592533 which links to this thread.

@Sineos — is this the same underlying issue? It seems that this issue keeps the heaters at their specified target, i.e. not reducing the targets to zero, whereas the linked issue had their HE temp almost double to > 450°C after an apparent Klipper error.

Also, you mention in the linked issue that GitHub is not the place for support questions, but does a Klipper error leading to thermal runaway not qualify as a significant development concern? 

Aside from obvious user error, one would think that Klipper should fail safe rather than forcing the heaters on an in uncontrolled manner. Just voicing my concern that this potential serious safety issue seems overlooked.

kjkent, 2023-09-24:
@KevinOConnor apologies for tagging you directly, I was just looking over my old commented issues that remain open, and https://github.com/Klipper3d/klipper/issues/5430 has stuck in the back of my mind since reading it earlier this year -- it seems @jdanders had a thermal runaway but the issue was closed as being not development-related, which I posit was an erroneous move.  I understand this is a separate issue, but @Sineos linked this issue in marking it 'not-for-github'. The linked issue, however, had the hotend continue to increase in temp rather than remain at its target.

Thank you for your time/consideration.

jdanders, 2023-09-24:
Hi @kjkent, to provide more context, the full diagnosis was done here on discourse: https://klipper.discourse.group/t/how-do-i-figure-out-why-mcu-was-disconnected/2634

The final conclusion was that the MCU overheated. I don't know MCU code well enough to know what's possible in that situation, but there are some comments about watchdog and scheduler code gaps. Once I hardwired the MCU cooling fan to "always on"  the problem has not happened again. 

Thanks for raising the issue again -- if an improvement can be made that prevents this it could save someone's house!  🔥

kjkent, 2023-09-25:
@jdanders thank you for this -- I really appreciate it! I'm currently learning embedded C but I don't know enough either to add any value to the discussion, but I'll definitely look through it, thank you.

KevinOConnor, 2024-01-23:
FYI, the functionality to disable heaters (or take other actions) was added in PR #5491.

-Kevin

jdanders, 2024-02-27:
@KevinOConnor do you see PR #5491 as a solution to this thermal runaway case? This was fixed by making sure the MCU cooling fan is always on, so I assume it's a case of losing the MCU entirely when it overheated. My [earlier comment](https://github.com/Klipper3d/klipper/issues/3666#issuecomment-1732703605) has all the details. 

The error in virtual_sdcard feature wouldn't work on that case, right? If not, is there any solution possible for this sort of error?

closed 2024-04-10 by github-actions[bot]