Klipper3d/klipper #3669: make menuconfig not working

Zeus1605, 2020-12-23:
Hello I am trying to install klipper to a skr pro V1.1
I followed the "Installation.md" but when running the make menuconfig command this is the result that I get. 
I am using a Raspberry Pi 4 with Octoprint V 1.5.2 and Octopi V 0.17.0
```
pi@octobig:~/klipper $ make menuconfig
Using default symbol values (no '/home/pi/klipper/.config')
Configuration saved to '/home/pi/klipper/.config'
  Creating symbolic link out/board
Loaded configuration '/home/pi/klipper/.config'
Traceback (most recent call last):
  File "lib/kconfiglib/menuconfig.py", line 3281, in <module>
    _main()
  File "lib/kconfiglib/menuconfig.py", line 661, in _main
    menuconfig(standard_kconfig(__doc__))
  File "lib/kconfiglib/menuconfig.py", line 705, in menuconfig
    locale.setlocale(locale.LC_ALL, "")
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
make: *** [Makefile:112: menuconfig] Error 1
```
Thanks Noah

closed 2020-12-24 by Zeus1605

lhess, 2021-02-06:
@Zeus1605 Why did you close this issue? How did you solve it?

I am having the same issue and it was somehow introduced with 0d5b05c704d1e15de2ce075f70d0de9f29ba1386. In [this file](https://github.com/KevinOConnor/klipper/blob/master/lib/kconfiglib/menuconfig.py#L705) `locale.setlocale(locale.LC_ALL, "")` is called and causes the following error:

```
Traceback (most recent call last):
  File "lib/kconfiglib/menuconfig.py", line 3281, in <module>
    _main()
  File "lib/kconfiglib/menuconfig.py", line 661, in _main
    menuconfig(standard_kconfig(__doc__))
  File "lib/kconfiglib/menuconfig.py", line 705, in menuconfig
    locale.setlocale(locale.LC_ALL, "")
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
make: *** [Makefile:112: menuconfig] Error 1
```

If I replace the empty string with something which makes more sense (like "en_US.UTF-8") it works... maybe it makes sense to set the value from the the env-var `LANG` which is usually set in any environment...

Zeus1605, 2021-02-06:
@lhess I actually didn't really fix the issue.
The error only happened when using gnome terminal so I tried using Putty instead and it worked with that so since then I just used Putty to install Klipper.
Thanks for the answer.

lhess, 2021-02-06:
Now I had a little bit more time to check the manual: https://docs.python.org/2.7/library/locale.html

it says for if an empty string is used:
> This sets the locale for all categories to the user’s default setting (typically specified in the LANG environment variable). If the locale is not changed thereafter, using multithreading should not cause problems.

Weird...

lhess, 2021-02-06:
Here the solution: https://stackoverflow.com/questions/36394101/pip-install-locale-error-unsupported-locale-setting

The fluidd base image which I am using for my pi has not the `LC_ALL` env var set...

lhess, 2021-02-06:
Okay, apparently its related to the terminal client that you use on your host system (for me its MacOS with iterm2). It sends the locale information when you ssh into a another machine. 

See:
- https://askubuntu.com/a/778672
- https://gitlab.com/gnachman/iterm2/-/issues/5478

hackaprende, 2021-03-24:
Same here, it happens with visual studio code terminal in MAC Big Sur but not in the default terminal.

guysoft, 2021-09-16:
Workaround I found was on OctoPi:

```
export LC_ALL=en_GB.UTF-8
make menuconfig
```

yawaweb, 2021-11-03:
> Workaround I found was on OctoPi:
> 
> ```
> export LC_ALL=en_GB.UTF-8
> make menuconfig
> ```

This works for me. Thank you!

redpiyaya, 2021-11-21:
This is a common issue when doing an SSH using a Mac Terminal. I tried using putty in Windows and it works fine. No error at all. I don't know why this happens but this is a FIX!