Klipper3d/klipper #3673: Use TMC5160 / TMC2130 in UART instead of SPI mode?

Tomblarom, 2020-12-24:
Hey!

I got a few TMC5160 for free, but these were the early prototypes and have nothing is broken out. Therefore I'm not able to active SPI-mode ([see](https://learn.watterott.com/silentstepstick/pinconfig/tmc5160/#operating-modes)). UART is enabled by default and no hardware modifications are required. Is it possible to use TMC5160 (and TMC2130?) in UART mode?

Tried setting `uart_pin` instead of `cs_pin`, but [klippy.log](https://github.com/KevinOConnor/klipper/files/5739797/klippy.log) throws this error:
`Option 'cs_pin' in section 'tmc5160 stepper_x' must be specified`

KevinOConnor, 2020-12-24:
The tmc2130 does not have the hardware support for uart mode and thus it can not be done.

The tmc5160 does support uart mode, but it hasn't been programmed in Klipper.  If you have the hardware, have it wired, and are willing to test it, I can probably put together some test code.  If so, let me know.

-Kevin

KevinOConnor, 2020-12-24:
On a closer look, the tmc5160 may not support UART mode using the standard STEP/DIR interface.  It's hard to tell from the specification.  You could give it a try, but I suspect you'll have better luck wiring the SPI lines and using that mode.

-Kevin

Tomblarom, 2020-12-24:
I got no documentation or pinout of my TMC5160. Watterott suspects my design is similar to [v1.3 / v13](https://github.com/watterott/SilentStepStick/tree/master/hardware). I was falsely assuming that UART is enabled by default and hoped to use UART instead. I asked the support and they ensure that SPI is enabled by default:
https://learn.watterott.com/silentstepstick/pinconfig/tmc5160/#operating-modes

[image]

Using `DUMP_TMC stepper=stepper_x` I'm getting the following output, which made me think, that there might be an issue, as you [mentioned in this guide](https://github.com/KevinOConnor/klipper/blob/master/docs/Sensorless_Homing.md#testing-of-spiuart-communication).
```
Recv: // ========== Write-only registers ==========
Recv: // COOLCONF:   000a0000 sgt=10
Recv: // GLOBALSCALER: 00000043 GLOBALSCALER=67
Recv: // IHOLD_IRUN: 00061f13 IHOLD=19 IRUN=31 IHOLDDELAY=6
Recv: // PWMCONF:    c40c001e PWM_OFS=30 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
Recv: // TPOWERDOWN: 0000000a TPOWERDOWN=10
Recv: // ========== Queried registers ==========
Recv: // GCONF:      00000000
Recv: // CHOPCONF:   00000000 MRES=0(256usteps)
Recv: // GSTAT:      00000000
Recv: // DRV_STATUS: 00000000
Recv: // FACTORY_CONF: 00000000
Recv: // IOIN:       00000000 VERSION=0x0
Recv: // LOST_STEPS: 00000000
Recv: // MSCNT:      00000000
Recv: // MSCURACT:   00000000
Recv: // OTP_READ:   00000000
Recv: // PWM_SCALE:  00000000
Recv: // PWM_AUTO:   00000000
Recv: // TSTEP:      00000000
Recv: ok
```

Tomblarom, 2020-12-25:
I had some FYSETC TMC2130 v1.0 ([solder-bridges set to SPI mode](https://reprap.org/wiki/TMC2130#Fysetec)) laying around and tested if they work as expected.. Unfortunately not, same as in https://github.com/KevinOConnor/klipper/issues/2416, https://github.com/KevinOConnor/klipper/issues/1509 and https://github.com/KevinOConnor/klipper/issues/1368..

Even temporarily swapped the SKR 1.4T for an identical one, I had on hand. Tried USB instead of 3-wire serial. Tried disabling TMC2209, because I thought they may interfer, due to this unusual jumpering https://github.com/KevinOConnor/klipper/issues/3672..

I don't know what to test else?

Tomblarom, 2020-12-27:
I was able to track my issue down to **two** broken TMC2130 🤦🏻‍♂️🤦🏻‍♂️ Tested one by one. It was also important to have [software spi pins defined](https://github.com/KevinOConnor/klipper/issues/1368#issuecomment-485916886).

From what I read, having issues like that, is always a configuration / hardware issue and has nothing to do with klipper.

@KevinOConnor I'd like to test the original topic UART on TMC5160, once I got new TMC5160s, since this may interest others in the future. Can't use the prototypes for this. I'd leave this issue opened due to that.

closed 2021-01-25 by github-actions[bot]