Klipper3d/klipper #3682: Error when compiling firmware on Raspberry Pi Zero W

Xkonti, 2020-12-26:
When following the klipper installation process the firmware compilation step fails. After sending following the command to the Raspberry Pi the process ends with an error:

`make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0`

Console output:

```
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
src/i2ccmds.c:14:23: error: field ‘i2c_config’ has incomplete type
     struct i2c_config i2c_config;
                       ^
src/i2ccmds.c: In function ‘command_config_i2c’:
src/i2ccmds.c:23:23: warning: implicit declaration of function ‘i2c_setup’ [-Wimplicit-function-declaration]
     i2c->i2c_config = i2c_setup(args[1], args[2], addr);
                       ^
src/i2ccmds.c: In function ‘command_i2c_write’:
src/i2ccmds.c:35:5: warning: implicit declaration of function ‘i2c_write’ [-Wimplicit-function-declaration]
     i2c_write(i2c->i2c_config, data_len, data);
     ^
src/i2ccmds.c: In function ‘command_i2c_read’:
src/i2ccmds.c:48:5: warning: implicit declaration of function ‘i2c_read’ [-Wimplicit-function-declaration]
     i2c_read(i2c->i2c_config, reg_len, reg, data_len, data);
     ^
make: *** [Makefile:64: out/src/i2ccmds.o] Error 1
```

There are no problems when connecting the OctoPrint with the printer.

Platform: Raspberry Pi Zero W
System: OctoPi 0.17.0
Printer: Anycubic Mega Zero 2.0

jasonobrien, 2020-12-27:
The issue is that `make menuconfig` incidentally sets up the `out/board` symlink before opening the menu where you'd select the architecture (so it just appends an empty string to the source root):

```shell
$ ls -al out/
lrwxrwxrwx  1 pi pi   27 Dec 27 00:45 board -> /home/pi/Code/klipper2/src/ 
```

The makefile isn't set up so as to automatically regenerate it when the config changes, but the easy fix is to just run `make clean` and `make` again.

chappy1978, 2020-12-31:
Thanks for the fix. I had the same problem on my Pi 3. Maybe you could include "make clean" in the setup instructions. So people don't get discouraged when they are setting up for the first time.

closed 2021-01-02 by klipper-gitissuebot

Hello1024, 2021-01-26:
This is still an issue with the latest version.  This should be reopened  @klipper-gitissuebot

Zeke13, 2021-02-20:
> 
> 
> Thanks for the fix. I had the same problem on my Pi 3. Maybe you could include "make clean" in the setup instructions. So people don't get discouraged when they are setting up for the first time.

Yes, adding this to the setup instructions would be awesome. I was getting frustrated trying to figure out what was wrong until I stumbled upon this thread.

oscarspaz, 2021-02-24:
I just run into the same issue. Thanks @jasonobrien for the instructions. 
This not a klipper-host issue. There is no klipper.log file. The console output from make is the error log.

burabure, 2021-03-16:
had this issue with latest release. you should really put this in the install guide

redfox05, 2021-04-27:
Still an issue as of 27 April 2021, with Version: v0.9.1-450-g475d8a72

The `make clean` suggestion fixed the problem.

Would recommend this be added to the docs at `https://www.klipper3d.org/Installation.html` under `**Building and flashing the micro-controller**`

dominickleppich, 2021-05-25:
Issue still present and undocumented. Luckily I found this :)

redfox05, 2021-06-16:
> It appears the Klipper log file was not provided.
I'm closing this now, sorry. Feel free to attach the log to this ticket sometime in the future, or create a new issue once you can provide the log file.

Based on the bot message above, this issue might get re-opened if a Klipper log file is added to the thread.
I'm not able to do that right now, but if someone else could, this might get re-opened?

brycestejskal, 2021-06-18:
Ran into this twice now. Here is your klippy.log @klipper-gitissuebot 
[klippy.log](https://github.com/KevinOConnor/klipper/files/6678276/klippy.log)

TheFanatr, 2021-06-22:
I PR'd the `make clean` "fix" to this, so that people know what to do when following the installation docs to get the firmware to build properly. Someone should still fix the `menuconfig` script if that's what the root problem is; not familiar with the Make makefile language myself, otherwise I'd do it. If I ever need to learn Make in the very near future and someone still hasn't fixed it, I'll go back and do it properly.

brycestejskal, 2021-06-22:
@TheFanatr - Thank you!

philipLutz, 2021-11-24:
This is still a problem. @TheFanatr where is the PR?

brycestejskal, 2021-11-24:
> This is still a problem. `menuconfig` should not work like this.

It is a minor bug.
Most people never see it.
The workaround is easily discovered and employed.

A few people run into it on a clean install, and they are typically able to resolve it with a quick Google search, leading them here.

Per the referenced PR, they may be changing directions here anyway.

Certainly not something high priority. It is debatable if it even worth noting in the docs for fresh installs.