Klipper3d/klipper #3683: Print fails after another print succeeded

Larvae90, 2020-12-26:
Hi,

I recently installed klipper and it's mostly working fine now that I'm done konfiguring. However I currently have a problem running multiple print jobs.

After the first job is finished and the printed part is removed I wanted to start a second print job. But every time I get a 'Must home before...' error, and the print is aborted.

The problem is that, like all my gcodes, the print job starts with a G28 for homing, which is executed correctly. The error occurs after homing and only after a print successfully finished (so far not after aborting a print). I tried a restart, but the error happended again. So I turned off the power to both printer and pi and restarted everything. Once thats done the print runs.

This happened to me 4 times already, every time with different parts being printed, every time it's the second print that fails, every time it works only after powering down completly.

Larvae90, 2020-12-27:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5744389/klippy.log)

Arksine, 2020-12-27:
Unfortunately your log does not show the error.  It would be helpful to reproduce this issue then upload the resulting log, otherwise it is difficult to determine the source of the error.  It would also be a good idea to issue an M112 after the error  so Klipper logs some relevant debug data.

Larvae90, 2020-12-27:
[Uploading klippy.log…]()
The log does contain the error, just without the M112. This new log contains a M112 send right after the error occurred.

Arksine, 2020-12-27:
I believe the upload for your log with M112 failed.  Apparently I missed the error the first time in the original log, I see it now.

Regardless, I think your issue could potentially be related to the call to FORCE_MOVE in your [homing_override] section.  It isn't necessary to force move the z stepper in a homing_override, by default the Z position is set to 0.  You can uncomment `G1 Z10 F600` to move the Z axis up 10mm.

Larvae90, 2020-12-28:
I can't replace theFORCE_MOVE. It's to lift the nozzle before deploying the probe when homing. G1 causes a "Must home before" error. Besides, as long as there wasn't a successful print before this code works fine. It's only when after a succsessful print that G28 fails after homing x/y and when homing z.

I just tried a G28 using the terminal. It fails the same way after a print as a gcode file.

Arksine, 2020-12-28:
Yes, you can and should replace the FORCE_MOVE.  See the [config refrence](https://www.klipper3d.org/Config_Reference.html#homing_override), specifically the `set_position_z` option.  

What is likely happening here is that bed_mesh is active and attempting to adjust the Z axis on when you move to (112.5, 112.5) in the homing override.  Setting the z position to 0 in the homing_override would resolve this.  Clearing the mesh in your end gcode using BED_MESH_CLEAR would also resolve this and is recommended.

Larvae90, 2020-12-28:
I think I missunderstood the `set_position_z` option when I first read it. With this, G1 works.

With that and BED_MESH_CLEAR it seems to work fine now. Thanks you very much

closed 2021-01-27 by github-actions[bot]