Klipper3d/klipper #3684: BTT skr mini e3 v2 config error

nleehone, 2020-12-26:
I'm getting the error ```NoOptionError: No option 'microsteps' in section: 'tmc2209 stepper_x'``` with the [btt skr mini e3 v2 config](https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v2.0.cfg).

I thought that the tmc2209 stepper_x would inherit the microsteps from the stepper_x (see attached log file). How can I resolve this?

```===== Config file =====
[stepper_x]
step_pin = PB13
dir_pin = !PB12
enable_pin = !PB14
microsteps = 16
rotation_distance = 40
endstop_pin = ^PC0
position_endstop = 0
position_max = 235
homing_speed = 50

[tmc2209 stepper_x]
uart_pin = PC11
tx_pin = PC10
uart_address = 0
run_current = 0.580
hold_current = 0.500
stealthchop_threshold = 250

[stepper_y]
step_pin = PB10
dir_pin = !PB2
enable_pin = !PB11
microsteps = 16
rotation_distance = 40
endstop_pin = ^PC1
position_endstop = 0
position_max = 235
homing_speed = 50

[tmc2209 stepper_y]
uart_pin = PC11
tx_pin = PC10
uart_address = 2
run_current = 0.580
hold_current = 0.500
stealthchop_threshold = 250

[stepper_z]
step_pin = PB0
dir_pin = PC5
enable_pin = !PB1
microsteps = 16
rotation_distance = 8
endstop_pin = ^PC2
position_endstop = 0.0
position_max = 250

[tmc2209 stepper_z]
uart_pin = PC11
tx_pin = PC10
uart_address = 1
run_current = 0.580
hold_current = 0.500
stealthchop_threshold = 5

[extruder]
step_pin = PB3
dir_pin = !PB4
enable_pin = !PD2
microsteps = 16
rotation_distance = 33.500
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PC8
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PA0
control = pid
pid_kp = 21.527
pid_ki = 1.063
pid_kd = 108.982
min_temp = 0
max_temp = 250

[tmc2209 extruder]
uart_pin = PC11
tx_pin = PC10
uart_address = 3
run_current = 0.650
hold_current = 0.500
stealthchop_threshold = 5

[heater_bed]
heater_pin = PC9
sensor_type = ATC Semitec 104GT-2
sensor_pin = PC3
control = pid
pid_kp = 54.027
pid_ki = 0.770
pid_kd = 948.182
min_temp = 0
max_temp = 130

[heater_fan nozzle_cooling_fan]
pin = PC7

[fan]
pin = PC6

[mcu]
serial = /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

[static_digital_output usb_pullup_enable]
pins = !PA14

[board_pins]
aliases = 
	EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
	EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>
=======================
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 146, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 139, in _read_config
    self.load_object(config, section_config.get_name(), None)
  File "/home/pi/klipper/klippy/klippy.py", line 128, in load_object
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/tmc2209.py", line 94, in load_config_prefix
    return TMC2209(config)
  File "/home/pi/klipper/klippy/extras/tmc2209.py", line 72, in __init__
    mh = tmc.TMCMicrostepHelper(config, self.mcu_tmc)
  File "/home/pi/klipper/klippy/extras/tmc.py", line 268, in __init__
    mres = config.getchoice('microsteps', steps)
  File "/home/pi/klipper/klippy/configfile.py", line 69, in getchoice
    c = self.get(option, default, note_valid=note_valid)
  File "/home/pi/klipper/klippy/configfile.py", line 55, in get
    note_valid=note_valid)
  File "/home/pi/klipper/klippy/configfile.py", line 32, in _get_wrapper
    v = parser(self.section, option)
  File "/usr/lib/python2.7/ConfigParser.py", line 340, in get
    raise NoOptionError(option, section)
NoOptionError: No option 'microsteps' in section: 'tmc2209 stepper_x'
webhooks client 3045581208: New connection
webhooks client 3045581208: Client info {'program': 'Moonraker', 'version': 'v0.1.0-132-gf96fb99'}
```

Sineos, 2020-12-26:
The microstepping setting belongs to the [tmc ....] section and not to the [stepper ....] section.

Sineos, 2020-12-26:
Edit: Changed for the latest Klipper version. Maybe you run an old copy. Try updating Klipper

nleehone, 2020-12-26:
I am on v0.9.1-61-g5773654a, so pretty recent.

The documentation states that it should be in the stepper section, not the tmc section: https://www.klipper3d.org/Config_Reference.html#stepper

BlackStump, 2020-12-27:
that is too old update klipper to the [latest version](https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software)
[config changes](https://www.klipper3d.org/Config_Changes.html)

nleehone, 2020-12-27:
Yes, you're right, it did work after updating to the latest.

I've reported the issue to mainsail so that they can update their packaged version or at least have a comment in the readme that will ask users to upgrade as soon as they install: https://github.com/meteyou/mainsail/issues/151

They will be releasing a new version soon which should deal with this issue.

closed 2020-12-27 by nleehone