Klipper3d/klipper #3686: How do I set up tmc2209 uart in the klipper?

cheshire7272, 2020-12-27:
Hi!

I have two-trees's sapphire plus v3 and upgraded to MKS Robin nano v2 and tmc2209 to use uart.

But there was a problem.

My printer has two Z-axes, but one of them doesn't work.

I checked the firmware, but I couldn't find the cause, so where should I fix it?

[stepper_x]
step_pin: PE3
dir_pin: !PE2
enable_pin: !PE4
step_distance: 0.01245
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_endstop: 0
position_max: 300
homing_speed: 60
homing_retract_dist: 0

[tmc2209 stepper_x]
uart_pin: PD5
microsteps: 16
interpolate: True
run_current: .5
hold_current: .5
sense_resistor: 0.110
stealthchop_threshold: 5000
diag_pin: PA15
driver_SGTHRS: 64

[stepper_y]
step_pin: PE0
dir_pin: !PB9
enable_pin: !PE1
step_distance: 0.01245
endstop_pin: tmc2209_stepper_y:virtual_endstop
position_endstop: 300
position_max: 300
homing_speed: 60
homing_retract_dist: 0

[tmc2209 stepper_y]
uart_pin: PD7
microsteps: 16
interpolate: True
run_current: .5
hold_current: .5
sense_resistor: 0.110
stealthchop_threshold: 5000
diag_pin: PA12
driver_SGTHRS: 64

[stepper_z]
step_pin: PB5
dir_pin: !PB4
enable_pin: !PB8
step_distance: 0.00125
endstop_pin: !PA11
position_endstop: 0
position_max: 340

[tmc2209 stepper_z]
uart_pin: PD4
microsteps: 16
interpolate: True
run_current: .5
hold_current: .5
sense_resistor: 0.110
stealthchop_threshold: 5000

[stepper_z1]
step_pin: PA6
dir_pin: !PA1
enable_pin: !PA3
step_distance: 0.00125

[tmc2209 stepper_z1]
uart_pin: PD8
microsteps: 16
interpolate: True
run_current: .5
hold_current: .5
sense_resistor: 0.110
stealthchop_threshold: 5000

[extruder]
step_pin: PD6
dir_pin: !PD3
enable_pin: !PB3
step_distance: .00243
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC3
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC1
control: pid
pid_kp = 17.039
pid_ki = 0.638
pid_kd = 113.732
min_temp: 0
max_temp: 280

[tmc2209 extruder]
uart_pin: PD9
microsteps: 16
interpolate: True
run_current: .5
hold_current: .5
sense_resistor: 0.110
stealthchop_threshold: 5000

[heater_bed]
heater_pin: PA0
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC0
control: pid
pid_kp = 75.754
pid_ki = 0.828
pid_kd = 1732.877
min_temp: 0
max_temp: 120

[fan]
pin: PB1

[mcu]
serial: /dev/ttyUSB0
restart_method: command

[printer]
kinematics: corexy
max_velocity: 200
max_accel: 3000
max_z_velocity: 25
max_z_accel: 100

[virtual_sdcard]
path: ~/.octoprint/uploads/

Tomblarom, 2020-12-27:
Who manufactured your TMCs? Are they in UART-mode?

Since you're using [sensorless homing](https://github.com/KevinOConnor/klipper/blob/master/docs/Sensorless_Homing.md), you must have read the guide on how to dump the tmc steppers. Right? That's one way to verify, if the UART connection is setup properly.

cheshire7272, 2020-12-27:
The tmc2209 driver is manufactured by MKS and is version 2.0.

All 5 drivers set jumpers in UART mode and sensorless homing function only allows X-axis and Y-axis.

> Who manufactured your TMCs? Are they in UART-mode?
>
> Since you're using [sensorless homing](https://github.com/KevinOConnor/klipper/blob/master/docs/Sensorless_Homing.md), you must have read the [...]

The tmc2209 driver is manufactured by MKS and is version 2.0.

All 5 drivers set jumpers in UART mode and sensorless homing function only allows X-axis and Y-axis.

cheshire7272, 2020-12-27:
Everyone, I'm sorry.

A mistake was detected and corrected while resetting the firmware.

Now my printer is working properly.

closed 2020-12-27 by cheshire7272