Klipper3d/klipper #3687: Printer Fail after Klipper update 1.3.1 bed temperature out of range

ZioBerny, 2020-12-27:
Hello,
after klipper upate all print fail, bet temperature goes out of range, below the klipper message:

MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

in attach the log file

Thank you
[klippy.log](https://github.com/KevinOConnor/klipper/files/5744278/klippy.log)

ZioBerny, 2020-12-27:
To add more info in attach the gcode and printer.cfg

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5744304/printer.cfg.txt)
[Optical-Sensor-new-asse.gcode.txt](https://github.com/KevinOConnor/klipper/files/5744305/Optical-Sensor-new-asse.gcode.txt)

ZioBerny, 2020-12-27:
Restarted all, resliced and seems working.
sorry for false issue

ZioBerny, 2020-12-27:
After some test, restart all many times the issue i raised again.
The print start:
heatbed reach initial temperature.
mesh bed leveling.
start to heat the hotend, at the same time the also the heatbed start to increase the temperature.
the error is raised.
I have checked the gcode and all seems ok, nothing is changed from last sucessfully print.

In attach Printer cfg and gcode
[Optical-Sensor-new.gcode.txt](https://github.com/KevinOConnor/klipper/files/5744457/Optical-Sensor-new.gcode.txt)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5744458/printer.cfg.txt)

ZioBerny, 2020-12-27:
I add also a screen capture, where you can see after the bed leveling all seems going fine, the heatbed temp is stable.
Once the bed level is done the "M109 S245" into gcode is sent to start to heat hotend/extruder, suddently the heatbed start to raise the temp also if the console say taget temp is 95, at the end once heatbed reach maximum temp the print stop with message:

**MCU 'mcu' shutdown: ADC out of range
This generally occurs when a heater temperature exceeds
its configured min_temp or max_temp.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown**

https://user-images.githubusercontent.com/74952641/103180072-a07e7800-4892-11eb-831f-a2d71c0020fb.mov

BlackStump, 2020-12-27:
Why do you think it is a Klipper issue?  Klipper seems to be doing exactly what it is suppose to which is to shut the system down because you have a runaway bed heater.
What I am seeing looks like a hardware/wiring issue.

ZioBerny, 2020-12-28:
Hi @BlackStump,
I think the issue could be related to Klipper, because nothingh else is changed on printer/sw, but not sure, yes, could be a hw  issue, I did some test, runing the commands from console:
PID_CALIBRATE HEATER=extruder TARGET=200
PID_CALIBRATE HEATER=heater_bed TARGET=60
and this commands working fine.
I have added all info like log, gcode, printer config to try to identify the issue. 
In any case, I'll change the title of Issue because no other people have raised the same issue, so now I'm pretty sure not related to the Klipper update.
Any suggestion to fix the issue is appreciate, I'll check the wiring and continue to try to understand the cause. 
Thank you for your help.

ZioBerny, 2020-12-28:
Issue solved adding following gcodes to slicer:

M140 S[first_layer_bed_temperature] ; set bed temperature
M104 S[first_layer_temperature] ; set temperature

Thank you

closed 2020-12-28 by ZioBerny