Klipper3d/klipper #3688: adding a tool locking stepper.

AmyTheCute, 2020-12-27:
how do I add a tool locking stepper? I couldn't find any documentations anywhere. I spent a lot of time and finally got it turned on and ready for the first time but I can't figure out how to make my tool locking stepper rotate!

any ideas?

AmyTheCute, 2020-12-27:
after HARD googling I found my answer to be manual_stepper

it'd be awesome to add it to example logs or documentations :) I couldn't find it ANYWHERE until I found the jubilee config files

closed 2020-12-27 by AmyTheCute

jakep82, 2021-01-04:
Every configuration option is listed in the official documents at klipper3d.org.  Manual stepper information is at the link below.

https://www.klipper3d.org/Config_Reference.html#manual_stepper