Klipper3d/klipper #3690: Fake Thermistor

guxtabo, 2020-12-27:
FAKE THERMISTOR TO 25C

Hi Kevin. I need to use my printer to print clay. This is why I don't use a heater or thermistor at all. How can I fake temperature sensor to always returning 25C?

KevinOConnor, 2020-12-27:
If you don't have an extruder at all then you should not define one in the config file.  If you do have an "extruder" but it does not have a heater then the simplest solution is to define the thermistor pin, but set min_temp=-273.15, min_extrude_temp=-273.15, and max_temp=99999999.

-Kevin

guxtabo, 2020-12-27:
> If you don't have an extruder at all then you should not define one in the config file. If you do have an "extruder" but it does not have a heater then the simplest solution is to define the thermistor pin, but set min_temp=-273.15, min_extrude_temp=-273.15, and max_temp=99999999.
> 
> -Kevin

I do have an extruder of clay, but not have a heater.
I did, but then I can't use any slicer, I can't input negative numbers into the slicer. That is why I ask you if you can create a thermistor that always measures 25C, even if it is not installed. (Fake thermistor).

guxtabo, 2020-12-28:
Kevin, can I fake the thermistor like this? To always sensing 25C, so I can use the slicer at 25C and no depends on the room temperature or not temperature at all?

[thermistor my_thermistor]
temperature1:25
resistance1: Resistance no thermistor
temperature2:25
resistance2: Resistance no thermistor
temperature3:25
resistance3: Resistance no thermistor

Arksine, 2020-12-29:
You could just hook a thermistor to the board, let it measure ambient, and set the `min_extrude_temp` to something below ambient.

guxtabo, 2020-12-31:
> You could just wire a thermistor to the board, let it measure ambient, and set the `min_extrude_temp` to something below ambient.

I will. Thank you.

guxtabo, 2020-12-31:
I wanted this: a dummy thermistor

https://github.com/KevinOConnor/klipper/issues/2300#issuecomment-567827266

closed 2021-01-30 by github-actions[bot]