Klipper3d/klipper #3692: Question about the current firmware-host protocol.

mfp20, 2020-12-27:
Still working on my [major update](https://github.com/KevinOConnor/klipper/issues/2683) idea. On firmware side of the story since june.

(TLTR) Long story short: I couldn't find some sort of "industry standard protocol"; basically to make an efficient protocol there's the need to specialize it and thread on the application peculiarities, so no general purpose protocol can match the 'custom ones'. Unless there's resource over-provisioning; but on AVRs isn't possible to waste any resource. About pulsing speed: Klipper slows down depending on the number of steppers; I was trying to keep the speed constant independently from the number of steppers, and the best results I could get was ... matching Klipper/GRBL speed. They are already the best in class on AVRs. So I went back to square 1: I'll use Klipper as a starting point. But I got a way better understanding of the firmware side of the story: AVRs may be EOL'ed, but are pretty good sparring partners... feel like Rocky Balboa running in the snow chained to a ton of wood logs. As an example a [COBS](https://pythonhosted.org/cobs/intro.html#intro) encoding might be [better than the current HDLC-like encoding](http://www.stuartcheshire.org/papers/COBSforToN.pdf) (ie: less overhead, there's a comparison to HDCL in the linked paper).

**I find all the DECL mechanism and the related build process pretty complex.** It works good, and I don't see any efficency problem. But the code complexity is a nightmare. Is it worth it? **What's the added value when compared to more conventional way to make an RPC system?** In other words why not just use a bunch of defines and structs to be included in both host and firmware? Something like:

```
// in .h

#define PKT_TYPE_UNKNOWN 0
#define PKT_TYPE_NOOP 1
#define PKT_TYPE_HELO 2
#define PKT_TYPE_STATUS 3
#define PKT_TYPE_INFO 4
#define PKT_TYPE_ACK 5
#define PKT_TYPE_NACK 6
#define PKT_TYPE_ECHO 7
#define PKT_TYPE_LOG 8
#define PKT_TYPE_SHELL 9
#define PKT_TYPE_PING 10
#define PKT_TYPE_OSP 11
#define PKT_TYPE_TOGGLE 12
#define PKT_TYPE_SENSE 13
#define PKT_TYPE_TICK 14
#define PKT_TYPE_STUFF 15

typedef struct status_s {
  uint8_t myid;
  uint8_t remoteid;
  uint8_t connection;
  uint8_t seq;        // current packet sequence index
  uint32_t tmicros;   // common time
  uint32_t tahead;    // action time
  uint8_t ahead_buffer_len;
  uint32_t pkts;
  uint32_t pktsError;
  uint16_t pulse_num;
} data_status_t;

typedef void (*pktHandlerPtr)(const uint8_t* buffer, size_t size);
pktHandlerPtr pktHandler[256];

void pkt_unknown(const uint8_t* buffer, size_t size);
void pkt_noop(const uint8_t* buffer, size_t size);
void pkt_helo(const uint8_t* buffer, size_t size);
void pkt_status(const uint8_t* buffer, size_t size);

// in .c, on init

pktHandler[PKT_TYPE_UNKNOWN] = pkt_unknown;
pktHandler[PKT_TYPE_NOOP] = pkt_noop;
pktHandler[PKT_TYPE_HELO] = pkt_helo;
pktHandler[PKT_TYPE_STATUS] = pkt_status;

// in .c, on packet rx:

pktHandler[rx_buffer[0]](&rx_buffer[1], size);
```

Then just build the same code for x86 and avr, and use the .h file as header. It can be both a python library and an obj for the firmware.

Am I missing something?

KevinOConnor, 2020-12-28:
A primary goal for the "compile time requests" and "data dictionary based mcu protocol" was to enable and encourage encapsulation.  Secondary goals were to support high efficiency and introspection for the protocol.

By encapsulation, I'm referring to the ability to keep like code close together, and to avoid having to maintain code that has global knowledge of all other parts of the system.  As an example of this, one can look at how the "neopixel" led support was originally added (commit b7bc96c3).  In that original commit, the neopixel C code and host python code was entirely added and enabled without any global "registration" code updates.  That update also automatically added support to [parsedump.py](https://www.klipper3d.org/Debugging.html#translating-gcode-files-to-micro-controller-commands) and the other debugging tools.  The same procedure was used when tmc_uart support was added (commit ca7a80a9), hd44780 lcd support (commit afc9bcf2), and many other features.

FWIW, I don't agree that the DECL_XXX system is particularly complicated.  For example, I think the addition of a DECL_COMMAND() line to declare a command handler is conceptuality easy to understand.  I think the high-level build process is also fairly straight forward (the build extracts the DECL_XXX() macros into out/compile_time_request.txt, scripts.buildcommands.py takes that file and produces out/compile_time_requests.c, and that C code is compiled into the final binary).  It is true that the build rules to extract the DECL_XXX() macros does require low-level knowledge of linker scripts, ELF file format, and their gcc interfaces.  However, I'm comfortable with these low-level interfaces, and wanted to make the code more approachable to those developers interested in higher-level concepts (eg, input shaper, pressure advance) and that may not be interested in low-level micro-controller development.

It is, of course, possible to obtain the above goals without the automated build scripts.  I suspect such a system would largely resemble the manual maintenance of something similar to out/compile-time-requests.c.  Alas, doing that would be tedious, error-prone, and be difficult to deploy some of the run-time optimizations.  FWIW, all these automated build scripts is notably less total lines of code than out/compile-time-requests.c.

YMMV.
-Kevin

mfp20, 2020-12-28:
Thanks (again) for your time. Existing docs are awesome but, as you stated about the code, it makes "the code more approachable to those developers interested in higher-level concepts", and I'm looking at lower level details instead. Without your help it would be much harder for me.

> A primary goal for the "compile time requests" and "data dictionary based mcu protocol" was to enable and encourage encapsulation. Secondary goals were to support high efficiency and introspection for the protocol.

I totally missed the 'introspection' goal; now I'm asking myself **what's the need of protocol introspection** (after development ended).

About 'efficiency' instead: overall speaking performance are higher than similar software, so Klipper talks by itself. I'm just pinning details here and I might be missing some more. But looking at the code for every command found by command_dispatch(), there is a call to command_parsef() to get the command details from progmem, before the command can finally by executed at the following func(args). In other words the **introspection costs some efficiency**; as usual, I'd say. 
**Am I wrong?**

> By encapsulation, I'm referring to the ability to keep like code close together, and to avoid having to maintain code that has global knowledge of all other parts of the system. As an example of this, one can look at how the "neopixel" led support 

I see the encapsulation point. But I'm currently not understanding, despite the examples, **what's the connection between DECL and encapsulation**. Need to read your words a second time and think about it a bit more.

> FWIW, I don't agree that the DECL_XXX system is particularly complicated.

**My bad. When I said 'nightmare' I referred to all the consequences of the DECL system; not the macro by itself.** The build system becomes pretty complicated by extra steps and details; that's the nightmare. Let's be clear: it isn't an issue, because all the extra steps are at build time. And once you did it, there's not much another developer have to do in order to take advantage of it. So, it's fine. But made my life pretty harder when I approached the code at first.

> It is, of course, possible to obtain the above goals without the automated build scripts. I suspect such a system would largely resemble the manual maintenance of something similar to [...]

I don't really know what alternatives exist. I've just investigated the need to include a C static lib into Python and got on web one of the current procedures to do it. Basically there's a 20 rows python script parsing all the .h to make a single .h file to be used at compile time. That single .h file has the same function of compile-time-requests.c, and it is automatically composed by the build system. In other words there's **no tedious maintenance, but it is very similar to your solution**. The only difference is that there's no need to go trough the extra steps for firmware building; it is needed for python's module build only.
Currently I don't know whether I can apply this solution to your code or not; I've setup this thing when I was building a firmware starting from blank file. And I've no idea if this can be any better. Probably just something that came out with Python3, after you incepted Klipper's project.

Anyway, **I got what I needed. Thanks.**

Q: What's the added value when compared to more conventional way to make an RPC system?
A: efficiency (?), introspection, and encapsulation (?).

I need to clear by myself the new question marks.

mfp20, 2020-12-28:
Another question looks to be in topic as well: why did you write in the docs "Unlike in HDLC, a sync character is not exclusive to the framing and may be present in the message block content." ?

You can't be sure to get some 0x7e (126) in the message block. Integers are mostly VLQ, but the 2 crc bytes could be equal to 126. The 'length byte' is forced down to 64; anything above that raise an error in command_find_block(); is that the only limiting factor for the MESSAGE_MAX ?

KevinOConnor, 2020-12-30:
> what's the need of protocol introspection (after development ended)

I don't envision a point where "development ends".  For example, adxl345 support was recently added (commit 3405095f), the neopixel support was changed to scale to more LEDs (commit 2bb30265), gpio updates are now queued in the micro-controller (commit 9cdf9bb6), etc.  Making the protocol more flexible, I think makes it easier to respond to changing requirements.  For example, in all of the above changes, users that updated and used one of the above features got a notice to flash their firmware again; users that upgraded but didn't use one of the above features were able to continue with the older version of the micro-controller code.

> But looking at the code for every command found by command_dispatch(), there is a call to command_parsef() to get the command details from progmem, before the command can finally by executed at the following func(args). In other words the introspection costs some efficiency

The introspection (specifically, the building of a data dictionary containing all supported messages and passing that information to the host) does not add run-time overhead.  The reason command_parsef() walks through all the parameters of each command is so that Klipper can implement [VLQs](https://www.klipper3d.org/Protocol.html#variable-length-quantities).  That is a useful compression technique which I estimate reduces total bandwidth roughly in half.  (And, reducing bandwidth itself reduces CPU usage because many micro-controllers need to take an expensive hardware interrupt on every byte read.)  In any case, I view the low-level message encoding as orthogonal to the "compile time requests" and "data dictionary based protocol" system.

> why did you write in the docs "Unlike in HDLC, a sync character is not exclusive to the framing and may be present in the message block content." ?

In HDLC, a 0x7e character may only be present in the header (as a sync character).  However, Klipper does not need nor enforce that requirement.  The document is pointing out that difference.

-Kevin

mfp20, 2020-12-30:
> > what's the need of protocol introspection (after development ended)
> 
> I don't envision a point where "development ends". 

My bad again. By 'development ends' I was referring to the point you don't need any more to analyze the running app in order to make the protocol more efficient. In general, yes, **I'm sold, it's good to have introspection**, maybe with a switch to include it at build time ("make development version")?
Probably the only needed change is to make the C modules a bit more traditional (ie: one .h each .c, using static and extern), then DECL commands in Hs rather than in Cs. But then encapsulation fades away... right... 

> > In other words the introspection costs some efficiency
> 
> The reason command_parsef() walks through all the parameters of each command is so that Klipper can implement [VLQs](https://www.klipper3d.org/Protocol.html#variable-length-quantities). 

It seems to me there's no need to walk trough parameters in order to implement VLQ. The standard VLQ contains in each byte 1 bit to tell whether the number ended or continue in the next byte; and your code seems to adhere the standard. So, again, I don't see the point of loosing efficiency in favor of introspection: cool for development purposes, bloatware in daily routine.

> That is a useful compression technique which I estimate reduces total bandwidth roughly in half. (And, reducing bandwidth itself reduces CPU usage because many micro-controllers need to take an [...]

I'm pretty sure VLQ is a must-have. I implemented VLQ in my own attempts as well before going back to your code. I just don't see the orthogonality with introspection.

> In HDLC, a 0x7e character may only be present in the header (as a sync character). However, Klipper does not need nor enforce that requirement. The document is pointing out that difference.

Yep, I got the point in the docs, but doesn't look consistent with code. Anyway, I've scrapped HDLC in favor of COBS. COBS recv algorithm need to parse the whole msg to remove the byte stuffing, this allows to find out the message length, so there's no need of the msg len byte at the beginning of every msg. And at the same time there's no chance for the sync code to be found in data bytes.

I'm close to get a running prototype and being able to run some tests about protocol efficiency. I'll let you know whether something good popped out or it is another fail.

Thanks again.

mfp20, 2021-01-03:
For the records: [current status](https://github.com/mfp20/klipper-firmware-ng). Any comment is welcome.

closed 2021-02-02 by github-actions[bot]