Klipper3d/klipper #3693: Printer shutdown during print with errors.

Alan5596, 2020-12-27:
Attached an HD webcam via USB to a RPI 3b and running 2 X SKR 1.3 on a Voron 2.4 with the Fluidd Web interface.

Before the webcam was added I had no printing errors and had completed about 10 prints.

I suspect the Webcam (Wansview 10180 HD) is the culprit but I have no experience of the Klippy.log so if anyone can see what caused it that would be great.  There were two errors - one just before the print started and one during the print.

"MCU 'z' shutdown: Missed scheduling of next digital out event"
"MCU 'z' shutdown: Timer too close"

Just finished the same print but with a low resolution cam and no issues.  Hopefully the cam was the problem.

[klippy (1).log.zip](https://github.com/KevinOConnor/klipper/files/5745118/klippy.1.log.zip)

KevinOConnor, 2020-12-30:
Both faults occurred because the klipper host software was unable to get any CPU time for multiple seconds.  The log also indicates that the "system load" on the host machine was exceptionally high (load of 8 at points).  So, it looks like something else running on the host machine resulted in the errors.

-Kevin

Alan5596, 2020-12-30:
Thanks for the reply.  
As a newbie to Voron and Klipper etc. I'm now puzzled as to what else may have caused it?  It's a standard build with SKR 1.3 2209's and Pi 3B and I'm not running anything else on the Pi apart from the webcam.
I have successfully completed two prints since the error occurred, one being the same print that failed so I can rule out any bad Gcode in the file.  Is there anything else I can monitor or look for that might be the cause?

I have updated Klipper, Moonraker and Fluidd yesterday so will keep an eye on the machine and see what happens.

Thanks for your help,

Alan

Alan5596, 2021-01-21:
Since removing the Webcam I have had no further problems so can ony assume it was the webcam or some other glitch in the system that caused the problem.

closed 2021-01-21 by Alan5596

Dragi2k, 2021-11-17:
I Have the same issue
Klippy log said :
Unable to issue reset command on MCU 'mcu'
Attempting MCU 'rpi' config_reset command
Got EOF when reading from device
webhooks client 1968783608: Disconnected
Restarting printer
Start printer at Tue Nov 16 21:38:06 2021 (1637095086.2 12499.2)

Moonraker said:
OSError: [Errno 11] Resource temporarily unavailable

And i have no idea why 😑