Klipper3d/klipper #3694: Disable G29 Automatic Backup

AntoRoni, 2020-12-28:
I would like to know how to disable the automatic backup of the Klipper configuration when I perform the G29
That looks like: printer-20201228_140030.cfg

AntoRoni, 2020-12-28:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5748801/klippy.log)

jakep82, 2020-12-28:
I'm not sure why I keep seeing people using save_config inside a G29 macro, but you really shouldn't have it there.  That will create a backup config file and restart the printer which makes it impossible to use G29 at the start of a print.

AntoRoni, 2020-12-28:
I am aware of this, but I am using G29 separate from Gcode, and I was wondering how to remove the automatic backup from SAVE_CONFIG

Bob90, 2020-12-30:
Not what you are asking directly, but I found this macro code:
BED_MESH_PROFILE SAVE=p1
So my guess would be to save a default profile you want to "P1" and load this to overwrite the default.

closed 2021-01-29 by github-actions[bot]