Klipper3d/klipper #3697: Sensorless homing not working [SKR v1.3 + TMC2209]

welo78, 2020-12-29:
Hi guys, I seem to be having issues with setting up the sensorless homing on Lerdge TMC2209. I switched to these drivers on X and Y but cannot get it to home at all. Previously I was using TMC2130 which worked just fine. I have a HyperCube CoreXY.

There are no jumpers present underneath the drivers themselves. UART jumper is in place as well as X and Y diag jumpers. Force move commands work flawlessly.

What I tried so far:
- numerous combinations on both stepper_x and stepper_y diag_pin with ! and ^ to no avail
- tried run current from 0.1 to 1 ( I read that this might be the cause of sensitivity issues)
- setting the sensitivity from 1 - 250 (setting it to 1 causes the motor to atleast move a few mms before triggering, if setup to 250 it does not move at all)
- combinations of all of the above

Klippy log might provide more information however the settings there are experimental, like I said I tried a lot of stuff.

Any suggestions are welcome. I for one think that the sensitivity even if set to 1 is still too high. Idk if there is an option to lower it even further or to use different diag pin ?

I am also adding the [official schematic. ](https://www.lerdge.com/public/upload/file/360f824755ab521bd741a67994b1de7e.pdf)

Below is my soldering/setup of these drivers. Force movement works flawlessly.
[image]
[image]

Klipper log:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5751137/klippy.log)

welo78, 2020-12-30:
It seems the issue is with the diag pin position. These drivers were created as Waterott equals and thus have the diag pin on the left side. However BTT TMC2209 have it on the correct side which is the right side. 

This means that a jumper cable must be in place in order to work properly. However using mechanical endstops and not using the stallguard/sensorless homing option at all could also fix this issue. I will come back with more findings.

dertester, 2020-12-30:
From the doc it should be right.
The diag is next to EN. Did you insert the jumpers and have nothing connected to corsseponding endstops?

Maybe your homing speed is to high / low?

welo78, 2020-12-30:
Which doc ? The skr board and klipper is configured the right diag socket. These Lerdge drivers are the same as Waterott which have the diag pin on left side. Big tree tech and fysetc TMC2209 have it on the right side and thus work from factory.

I did try a makeshit solution by soldering a long pin on the left and bending it to be on the right side in a "S"  ... all is working now flawlessly which supports my theory.

dertester, 2020-12-30:
Everything okay. Left and right are maybe not the best way for communication.

closed 2021-01-29 by github-actions[bot]