Klipper3d/klipper #3698: Random MCU Drop

telxoid, 2020-12-29:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5751502/klippy.log)

Hi, having random MCU drop issues and looking for a direction.  "Transition to shutdown state: Lost communication with MCU 'z'".  It doesn't seem to care if the printer is printing or idle.  What I've tried:

Environment is Voron 1.8 with 3 screws, so 2 SKR 1.3's with currently 3 steppers populated in each.  Raspi 3B that was running a different printer without issue.  Config is pirated from my running Voron 2.4 and mildly modified, so lots of comments ATM. 
What I've done so far to try and fix:
- Upgraded to the latest klipper on both my raspi and the MCU's (at that time I also changed bootloaders to allow for USB upgrades)
- Rebuilt raspi with new SD Card.

It seems like it is mostly (grepping through klippy logs) the MCU labeled as mcu z, but I've also noted a single instance of the other MCU dropping, so I'm hesitant to replace hardware.

Any clues?  Need more info?

Thanks!

Klaygar, 2021-01-10:
Hi.
Recently faced a similar problem, even thought that the control boards or Pi3B were damaged. I tried to change the USB flash drive and even replaced the control board, but the problem remained.

I tried 4 usb cables and only at the last one the work became stable, while all the cables are working and the problem appeared after some of the octoprint and klipper updates

telxoid, 2021-01-12:
Looks like I had a hardware issue.  Finally changed the raspi out of desperation and it stopped.

closed 2021-02-11 by github-actions[bot]