Klipper3d/klipper #3699: 60W Hotend Heater fluctuating

PerLangkjaer, 2020-12-30:
As You can see I have some problems with the new heater.
I have tried PID tuning at 215 degrees and 190 degrees and same problem

[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/5752964/klippy.log)

jakep82, 2020-12-30:
Very powerful heaters can overwhelm the PID loop.  Try adding `pid_integral_max` to your extruder section and see if a value lower than the default of 1.0 helps.

PerLangkjaer, 2020-12-31:
Thanks, I'll have a look at it when my next line of prints is done 👍 
I've tried to put Max_Power to less than 1.0 and about 0.5 it got controllable, at least in the software.
But the prints showed too much heat still.
It looked like the temperature went up and down so fast that the software could not see it but it cooked the filament none the less.

PerLangkjaer, 2021-01-02:
@jakep82 
Nope, adding the integral part dod not do anything good.
It actually caused the temperature to fluctuate even more :(
Tested from 0.8 to 0.5 and no good.
I have been playing with the thought of changing the PWM_SPEED or something like that 
Forcing the firmware to send a signal more often, but that requires the temp sensor to be read even more often....

PerLangkjaer, 2021-01-05:
After more testing with different setups in the config, I've decided to lay it down. I can't get it to work and have now moved back to a 40W that works perfectly.

jakep82, 2021-01-05:
Unless you're using a super volcano you likely don't need more than 40W.  You could probably get it to work with a very low max_power setting, but then you get no benefit over using a smaller 30 or 40 watt cartridge.  If you're using a standard heater block, it's very difficult to control that much power with such little thermal mass.

closed 2021-02-04 by github-actions[bot]