Klipper3d/klipper #3701: Skr 1.4T 2209 dual-z moves extruder

klexxp, 2020-12-30:
Hi
I have a weird problem with my setup. Whenever I move the z-axis, the extruder also moves in the same direction. When I try to extrude it wont move.

I was using Marlin with the same setup before. I checked the config of the pins a few times, but it seems all correct to me. 

Maybe someone got a hint...
Thx

[klippy (1).log](https://github.com/KevinOConnor/klipper/files/5753926/klippy.1.log)

dertester, 2020-12-30:
Hey,

do you have dual z with dual endstops?
You dont have a second z definded then! Its X,Y,Z1, E1, Z2 on your board.

I think you should be able to use my pins even if i use a skr 1.3. 

Here is what my conf looks like with dual z, Dual Endstop.

`[stepper_z]
step_pin: P0.22
dir_pin: !P2.11
enable_pin: !P0.21
step_distance: .005
endstop_pin: P1.25  # P1.24 for Z-max

homing_retract_dist: 15.0
second_homing_speed:5

position_endstop: 0
position_max: 350
position_min: -10

[tmc2209 stepper_z]
uart_pin: P1.10
microsteps: 32
run_current: 1.2
hold_current: 0.9
stealthchop_threshold: 70

[stepper_z1]
step_pin: P0.1
dir_pin: !P0.0
step_distance: .005
enable_pin: !P0.10
endstop_pin: P1.24  # P1.24 for Z-max

[tmc2209 stepper_z1]
uart_pin: P1.1
microsteps: 32
run_current: 1.2
hold_current: 0.900
stealthchop_threshold: 70`

klexxp, 2020-12-30:
I'm running sensorless homing on x and y and the bltouch on z, which I tested and is working. 
I have the second z stepper connected on E0 and the extruder on E1.

dertester, 2020-12-30:
Fist: You did not answer the question about second endstop.

To help with info provided:
Here you find the Pinout for SKR 1.4.
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf

As you can see in the diagram the E1 where your Extruder is has:
EN 1.16
Step 1.15
Dir 1.14

But your Config for extruder is:
step_pin = P2.13
dir_pin = !P0.11
enable_pin = !P2.12

That may be the problem! You got the pins wrong
Also you need to add a Stepper_z1 for your second stepper.
Follow my example in my last comment.

Actually its more common to leave E0 for Extruder and E1 for second Z. Its also easier in Marlin but i guess you want
to keep backward compatibility so you have to change pins.

You need to set up
stepper_x
stepper_y
stepper_z 
stepper_z1 (Pins for E0)
extruder (Pins for E1)

Good luck!

klexxp, 2020-12-30:
Yes, you are right. I thought that  I had connected second z on E0.
I forgot that the board has two connectors for the z-axis and that they share the pins. Was some time ago I hooked that up ;)

Thank you for taking the time to look into this.

closed 2020-12-30 by klexxp