Klipper3d/klipper #3702: Too much extrusion followed by under extrusion

arturohernandez10, 2020-12-30:
I just recently installed klipper and it is really cool.

While printing a benchy on an ender 3 with the silent board installed, with a BMG extruder on a direct extruder configuration. I have some extrusion issues. No matter what advanced pressure setting or printing speed I use I get the same random artifacts in random places. It's not retractions. I verified the retractions are not even close to that portion of the path. I isolated one example by cutting the STL so I would only print the cabin. On that gcode file it happens consistently at layer 10. So I cut the gcode file, verified it still happens. And I am including photos, the log file and the gcode file. I can tell it's doing heavy extrusion on the corner and then it does not. But it does it randomly. It took a little time to get these files together but I want to help.

[3DBenchyCabin9layers_plusPartial.zip gcode file](https://github.com/KevinOConnor/klipper/files/5755658/3DBenchyCabin9layers_plusPartial.zip)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/5755646/klippy.zip)

This is the cut gcode file so you can see the extrusion issue exactly where it happens. The arrow is in the direction of the extrusion.

[image]

This is the original side view. Where I saw the issue. This will happen in random places.

[image]

BlackStump, 2020-12-31:
do you have Z calibrated correctly and/or have you check your Z for binding?

arturohernandez10, 2020-12-31:
I think it is, I'll check it. I'm not sure how that could be related. This is happening 5 corners away from the z move. Do you have something in mind, or a preferred way to calibrate the z?

arturohernandez10, 2020-12-31:
Just as a test, since it is 100% consistent where the problem occurs. I copied the gcode for every layer to add 3 more identical to the one with the error. I added a 

     G92 E127.6652

gcode between layers so that I would not have to change the absolute extruder values from CURA. Also there is only one z move per layer. So it was easy to do.

I included a picture of the result. The issue only presents itself in the 10th and 11th layer. The 12th and 13th don't have that issue. They do have some internal line artifact by superimposing a line that I doubt CURA would like to be repeated on top of each other. 

I'm not sure what is the look ahead window. But this looks like a rounding error, that maybe carried around but eventually disappears. Because klipper is evidently printing the exact same gcode layer just fine. Just for reference there are 78 gcode commands with extrusion in each repeated layer, and I recall 3 retractions. The issue is between command 9 and 10.

[3DBenchyCabin10layers_plus3.zip gcode](https://github.com/KevinOConnor/klipper/files/5756191/3DBenchyCabin10layers_plus3.zip)
[image]

arturohernandez10, 2021-01-06:
In the extrusion amount image, I marked the amount of steps on darker color based on the klipper commands sent from klippy to the microcontroller. And also included the layer from octoprint for reference. The extrusion is less on the corners, which does not explain why I consistently get this artifact every time in the exact same layer. And this same kind of artifact appears in many other places when printing a benchy.

One correction is the direction of print. this is happening while moving the printer bed towards the back of the printer. The extrusion is less than what it should be. Then at the corner all the missing material comes out.  The previous post this is right to left.

Any advice will be greatly appreciated

[image]
[image]

KevinOConnor, 2021-01-24:
So, to summarize, if you duplicate the gcode on multiple lines, the problem does not occur?  And, when analyzing the klipper micro-controller commands, you don't see anything that looks incorrect?

If so, it doesn't seem like it's the software..

Could there be additional stress on the filament depending on head XY moves?

It's possible this could be a quirk of using stealthchop mode on the extruder.  Alas, I don't think the creality "silent board" allows one to easily disable stealthchop mode.

Do you have any further theories as to what it is?

-Kevin

closed 2021-02-19 by arturohernandez10