Klipper3d/klipper #3703: Unable to connect with mcu rpi

furkanbr, 2020-12-31:
Hi to everyone , im having problem with my printer for few days and i cant solve it.

My printer and klipper firmware is working without any issue and few days ago , i got an adxl345 accelerometer and start to prepare my firmware. i followed all steps in "Measuring Resonances" guide but klipper giving me an error as "Unable to connect". if i delete mcu rpi , adxl345 and resonance_tester sections from my cfg file , everythink works fine again. Can anybody help me? i cant solve this problem.

I triend ; different sd card , re installing octopi and klipper , updating all firmwares ...

My hardware ; Raspberry pi 4 4gb , skr mini e3 v2

[klippy.log](https://github.com/KevinOConnor/klipper/files/5756223/klippy.log)

arturohernandez10, 2020-12-31:
The new version of the firmware has a new serial scheme. I had to copy and paste from the console after flashing the board.

Sineos, 2020-12-31:
Your setting
```
[mcu]
serial = /dev/ttyACM0
```
looks wrong. Make sure the board is properly flashed then do `ls /dev/serial/by-id/*` to find the right serial.

furkanbr, 2020-12-31:
@Sineos

furkanbr, 2020-12-31:
@Sineos ls /dev/serial/by-id/* command gives me "usb-Klipper_stm32f103xe_35FFD8054247343133660957-if00" but when i tried to connect with this port , it can connect without mcu rpi , adxl345 and resonance_tester sections but if i add them to my cfg file , fimware still give me an error as unable to connect. 

@arturohernandez10 skr mini e3 does not support make flash command , im flashing it with sd card. and configuration is working fine if i dont try to use mcu rpi , adxl345 and resonance_tester sections. 

Thanks!

Sineos, 2020-12-31:
`usb-Klipper_stm32f103xe_35FFD8054247343133660957-if00` looks correct for your board.

For the RPI MCU be sure to follow https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md step by step. It looks like you didn't correctly build / install the RPI MCU.

furkanbr, 2020-12-31:
@Sineos @arturohernandez10 

Thanks for your helps !! its finally working and it was just my fault , i just missed 

``To compile the Klipper micro-controller code, start by configuring it for the “Linux process”:``

in the guide and make my firmware for my stm32 and it just forking awesome now , thanks to everyone!!

closed 2021-01-01 by furkanbr