Klipper3d/klipper #3704: Creality 4.2.7 board, unable to connect.

josephwunder, 2020-12-31:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5756414/klippy.log)

Hey everyone, 

I'm running an Rpi 4 and i have a creality 4.2.7 board. I could not flash via usb so i used the instructions at the top of the generic 4.2.7 config, set the mcu to STM32F103 with a "28KiB bootloader" and with "Use USB for communication" disabled. flash my firmware to an sd card. put it on the board. My screen goes solid blue. And my octopi setup says [Recv: ok FIRMWARE_VERSION:v0.9.1-140-ge68cf08d FIRMWARE_NAME:Klipper] So it sees there is firmware on the device. But even with my serial port set to /tmp/printer and baud rate to 250000, i cannot send any commands to the printer. My temperature keeps reading 0.

I've tried other usb cables, i tried reflashing the firmware and tried config files including the basic and one i found on https://github.com/KevinOConnor/klipper/pull/3335/commits/f02f24a088995bd789418423cd71a7f51dcf6a34 and i cannot get it to connect. any help would be appreciated. :)

jakep82, 2020-12-31:
From your log:

```
Unable to open port: [Errno 2] could not open port /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0'
```

You need to fix your config file.  After flashing the board, it likely changed names.  Run `ls /dev/serial/by-id/*` to find the correct by-id name.

josephwunder, 2020-12-31:
I'll give that a try, see how it goes, i could have sworn a ran the name again and the same port name showed up

closed 2021-01-02 by josephwunder