Klipper3d/klipper #3706: Drifting z Offset

oeijoenj, 2020-12-31:
I have situation when z offset is drifting for each print.
Just now I have 2.5 hrs printing done successfully. I was waiting until bed temp reach lower temp and initiate another print but failed in first layer.
Did z probe offset and z offset was drifting from 1.450 to 1.900.

A couple weeks later, I had same situation when z offset drifting from 2.700 to 3.100. I though it was due to yellow spring not fasten enough then I fastened the knob and that's why z offset changed to 1.450.

Any suggestion?

aerialviews007, 2021-01-02:
I'm having a similar issue but I'm fairly certain it's the probe.  Have you done a probe_accuracy test?

oeijoenj, 2021-01-02:
Yes, I did probe_accuracy.
I just did it again and here is the result 

Send: PROBE_ACCURACY
Recv: // PROBE_ACCURACY at X:117.000 Y:117.000 Z:10.000 (samples=10 retract=2.000 speed=5.0 lift_speed=5.0)
Recv: // probe at 117.000,117.000 is z=1.912500
Recv: // probe at 117.000,117.000 is z=1.917500
Recv: // probe at 117.000,117.000 is z=1.920000
Recv: // probe at 117.000,117.000 is z=1.920000
Recv: // probe at 117.000,117.000 is z=1.922500
Recv: // probe at 117.000,117.000 is z=1.922500
Recv: // probe at 117.000,117.000 is z=1.920000
Recv: // probe at 117.000,117.000 is z=1.922500
Recv: // probe at 117.000,117.000 is z=1.925000
Recv: // probe at 117.000,117.000 is z=1.920000
Recv: // probe accuracy results: maximum 1.925000, minimum 1.912500, range 0.012500, average 1.920250, median 1.920000, standard deviation 0.003250

Regarding to horizontal_move_z, any recommendation from you? Mine is 5 as well.

Thanks.

> I'm having a similar issue but I'm fairly certain it's the probe. Have you done a probe_accuracy test?
> 
> Edit: Also make sure your horizontal_move_z in bed mesh is big enough. Mine was originally 5 and it was not enough.

oeijoenj, 2021-01-03:
It just happened again.
I have tightened  the yellow spring and did screw_tilt calculate the day before.
Today I just finished print object with z offset 1.915 and and initiate another print but failed in first layer.
Did probe_calibrate with bed and hot end at 240 and 100 ( my working temp for ABS ) and got 2.200.

Initially, I though it due to loose spring but I dont think that is the case.
 
> Yes, I did probe_accuracy.
> I just did it again and here is the result
>
> Send: PROBE_ACCURACY
> Recv: // PROBE_ACCURACY at X:117.000 Y:117.000 Z:10.000 (samples=10 retract=2.000 speed=5.0 [...]

aerialviews007, 2021-01-03:
My horizontal Z is at 8 because my offset is a bit more than yours.  5 mm wasn't enough to clear the probe.

I'm at a loss.  My accuracy is nowhere near yours so I can't conclusively say we have similar issues.

LeviTheOtaku, 2021-01-05:
every test the z_offset just raised by about 0.01mm-0.02mm

```
Send: PROBE_CALIBRATE SPEED=50 SAMPLES=10
Recv: // probe at 118.000,118.000 is z=4.510000
Recv: // probe at 118.000,118.000 is z=4.517500
Recv: // probe at 118.000,118.000 is z=4.517500
Recv: // probe at 118.000,118.000 is z=4.522500
Recv: // probe at 118.000,118.000 is z=4.532500
Recv: // probe at 118.000,118.000 is z=4.532500
Recv: // probe at 118.000,118.000 is z=4.535000
Recv: // probe at 118.000,118.000 is z=4.540000
Recv: // probe at 118.000,118.000 is z=4.545000
Recv: // probe at 118.000,118.000 is z=4.550000
```

I'm using A4988 stepper drivers and a SKR 1.4 Turbo.

closed 2021-02-04 by github-actions[bot]