Klipper3d/klipper #3707: Creality Ender 3 Pro - 1.1.4 - Extruder Fan Pinout Confusion

crossan007, 2020-12-31:
I've recently installed Klipper on my otherwise "stock" Ender 3 Pro.

I followed the guide to flash the bootloader from here: https://www.fission3d.com/post/flash-bootloader-and-install-firmware-with-raspberry-pi  and then proceeded to follow the [Klipper installation guide](https://www.klipper3d.org/Installation.html)

My Ender 3 Pro has the Creality 1.1.4 controller board.

I used the sample config from here: https://raw.githubusercontent.com/KevinOConnor/klipper/master/config/printer-creality-ender3-2018.cfg

While following [the config check document](https://www.klipper3d.org/Config_checks.html), I also attempted to turn on the cooling fan, but noticed that the controller fan turned on instead of  the nozzle cooling fan

I did some searching for pinouts, and have been mostly unsuccessful at finding another fan control pin.

I located (what I assume to be) the best-matching pinout to my board [from the official Creality repository](https://raw.githubusercontent.com/Creality3DPrinting/Ender-3/master/Ender-3%20Firmware%20(Marlin)/Ender-3%20(includes%20power%20failure%20resume-%20English)/Marlin/pins_MELZI_CREALITY.h) ; however, the `FAN_PIN` referenced in this file seems to be the same pin as referenced in the `printer-creality-ender3-2018.cfg` file.

Where else can I look to figure out which pin my extruder nozzle fan connects to?

[klippy.log](https://github.com/KevinOConnor/klipper/files/5757649/klippy.log)

crossan007, 2020-12-31:
I found an Ender 3 Schematic, and it seems to indicate there's only one fan control port: https://github.com/RudolphRiedel/CR-10_wiring/blob/master/Ender3_schematic.PDF

crossan007, 2021-01-01:
I opened up the controller box again, and it seems that the controller fan and the hotend fan are driven by the same micro controller pin.   I confirmed that 24V flows to the hotend fan as expected when I send the Fan on command;  it seems that my hotend fan failed concurrently with flashing Klipper. 

Closing this issue as an extremely unlucky coincidence. 

Both fans are pin `PB4`

[image]

closed 2021-01-01 by crossan007

MoZoller, 2021-03-27:
I have exactly the same error after the same procedure