Klipper3d/klipper #3710: Implement Temporary Temperature Check Bypass

crossan007, 2021-01-01:
# Summary 
My printer is currently in my unheated garage in southeastern Pennsylvania: sometimes ambient temperatures will drop to ~0*C. or lower

This low ambient temperature can trigger the `min_temp` "thermal runaway" shutdown:
```
Recv: // MCU 'mcu' shutdown: ADC out of range
Recv: // This generally occurs when a heater temperature exceeds
Recv: // its configured min_temp or max_temp.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: ADC out of range
```

Before installing Klipper, I would "warm" the bed and nozzle with my hands (quite unpleasant!) until the measured temperature was within appropriate bounds.  Now, with Klippper, I can adjust my `printer.cfg` file to permit this lower temperature, but I feel this configuration change reduces the effectiveness of the thermal runaway protection feature.

# Feature Request

I would like to see an option for a "one-time bypass" of temperature checks so that I can heat my bed/nozzle.  Once the heating begins and the heater becomes "within range" of the permitted values, the temperature checks should resume.

This will allow "more safe" values to be specified in `heater_*` configurations for `min_temp` and `max_temp`.

# Related Issues
https://github.com/KevinOConnor/klipper/issues/3290 - Is there a way to disable temperature checks?

BlackStump, 2021-01-01:
I presume you have added min_temp: to your config?   but you have not provided a log as requested so only you would know.
[extruder]
min_temp: -5 # or whatever you min ambient temp is

[heater_bed]
min_temp: -5

crossan007, 2021-01-01:
@BlackStump yes, I have set `min_temp` appropriately (though, you're right; I should have included a log):
[klippy.log](https://github.com/KevinOConnor/klipper/files/5759569/klippy.log)

The spirit of this request is to allow setting `min_temp` within "normal operating bounds" and maintain close safety-tolerances during normal operation, while still facilitating a "system startup" from an out-of-bounds scenario

BlackStump, 2021-01-02:
fair enough sorry can not help in that department, I would imagine a developer would have to develop and submit that.

crossan007, 2021-01-02:
I've got some development experience, but I haven't looked at the Klipper source yet. 

If I get a sudden burst of inspiration I may take a shot at this and submit a merge request.  Otherwise, some other developer could probably implement this a lot quicker than I could ;-)

closed 2021-01-31 by github-actions[bot]