Klipper3d/klipper #3711: BLTouch does not work stable

ThomasW69, 2021-01-01:
Yesterday I got my new SKR 1.4 runninng because there was no communication to my MKS Gen L possible anymore. 
After some configuration problems everything was ok an I did some prints. 
Now suddenly my BLTouch did stop working while I was perfoming a bed mesh calibrate. So, in the second row of the mesh the nozzle crashed into the bed!  Then I checked the cables but all are OK. After switscing power on again there is no reaction to the pin_down, pin_up and reset commands for the BLTouch.  
[klippy.log](https://github.com/KevinOConnor/klipper/files/5759069/klippy.log)

ThomasW69, 2021-01-01:
Could this problem caused by the issue of the SKR1.4 Turbo board decribed in this video at 26.15?
[https://www.youtube.com/watch?v=p504oU-D6iE](https://www.youtube.com/watch?v=p504oU-D6iE)

Sineos, 2021-01-01:
I'm using a genuine BLTouch on a SKR 1.4 Turbo without problems. Wired it to the Z-Endstop pins as the dedicated BLTouch pins never worked for me (didn't spend too much time on it though).

ThomasW69, 2021-01-02:
mine is connected to the servo and probe pins of the board.

closed 2021-02-01 by github-actions[bot]