Klipper3d/klipper #3713: SKR mini e3 v2.0 - the skr is flashed but can't connect to the skr via !PA14

iDestz, 2021-01-02:
Hello, we have a problem that the skr mini e3 v2 does not turn on/unlock the usb port. 
After firmware.bin is successfully flashed (checked renaming in firmware.cur). The board is not visibly at the PI. Either by ls -l /dev/serial/* or lsusb
- We tried it now several times
- We used klipper.bin files generated from two different machines
- yes the GPIO pin !PA14 is set
- yes 28k is set
- yes STM32F103 is set
We tried also all 4 USB ports on the rpi with the lsusb command and sticked in usb mouse. We see the mouse on all Ports.
We test the Signal from the usb end of the blue skr cable to the skr usb port. v+,v-, d+ and d-. All ok.
Any idea?

thank you for tips

Dominik

iDestz, 2021-01-02:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5759792/klippy.log)

it is the wrong serial in the [mcu] section, because the usb port is not unlocked. i can't read with ls /dev/serial/by-id/* the right serial.

Sineos, 2021-01-02:
Your board is not even recognized by the operating system. Check with `dmesg` after connecting the board. 

If you do not receive anything like this
```
[1202456.203302] usb 1-1.3: USB disconnect, device number 36
[1202467.506222] usb 1-1.3: new full-speed USB device number 37 using dwc_otg
[1202467.641065] usb 1-1.3: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[1202467.641085] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1202467.641099] usb 1-1.3: Product: lpc1769
[1202467.641113] usb 1-1.3: Manufacturer: Klipper
[1202467.641127] usb 1-1.3: SerialNumber: 0C70001625813AAF86E06B5CC32000F5
[1202467.642424] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
```
then something is dead. Either the USB on the host or on the board.

Roycinger, 2021-01-02:
1) it's by-id, not byid
2) did you make a make clean before the make?
3) war the firmware flashed properly (firmware.cur on the sd card after power cycling the board)?

Arksine, 2021-01-03:
Make absolutely sure that you have a Mini V2 and not a V1.2.  They have different USB Discovery pins, v1.2 uses PC13.

iDestz, 2021-01-03:
> Your board is not even recognized by the operating system. Check with `dmesg` after connecting the board.
>
> If you do not receive anything like this
>
> ```
> [1202456.203302] usb 1-1.3: USB disconnect, [...]

Entered the command and the board was displayed. Then I took the serial number and entered it into the printer.cfg and restarted the rpi. Then I entered the command again. The board was then no longer displayed. Now I have the third cable through. With all three, the board is no longer displayed. Then the board must be defective

iDestz, 2021-01-03:
> Make absolutely sure that you have a Mini V2 and not a V1.2. They have different USB Discovery pins, v1.2 uses PC13.

I am absolutely sure that I have the skr mini e3 v2. That's what it says on the board. It's not mislabeled either because I have the two z-motor slots on the board. Also matched with the pinout from btt

iDestz, 2021-01-03:
> * it's by-id, not byid
> * did you make a make clean before the make?
> * war the firmware flashed properly (firmware.cur on the sd card after power cycling the board)?

1) I know. Was also still pointed out that I have it wrong in the screenshot. I had logged in again via ssh when I wrote the issue and wanted to attach a screenshot. In doing so, I entered the command manually. But I am 100% sure that I copied the commands from the manual correctly before. So /by-id/*

2) Yes, every time.

3) Yes, the firmware was flashed successfully and on the sd card was after flashing the firmware.cur. 

Generally I checked the usb ports on the rpi. All ports recognize usb-mouse. I measured the cable by plugging the mini usb cable into the board and measured behind the mini usb port. Both data ports beep, so no cable break. Nevertheless I tried three cables. Someone in the community made me a firmware from his pc and I would say he has in depth knowledge. I have received a firmware.bin from someone who has the same setup. rpi4 + skr mini e3 v2. 

Funny thing was that after the first make with ls I queried /dev/serial/by-id/* and also got a serial. There was something with "leaf" and "Maple". After that it never recognized the serial again until the tip from above where I should enter the command. There it showed me the serial where it should be under Klipper. Something like: stm32....xe....klipper_xxxxxxx (xxxx are numbers of course). After that I copied the serial and entered it into the Printer.cfg and rebooted the rpi. Apparently the usb ports were queried again and it was not displayed again. I guess that the board is broken somewhere or has a loose contact on the board.

iDestz, 2021-01-03:
Last login: Sat Jan  2 05:15:55 on ttys000
beck@Dominiks-MacBook-Pro ~ % ssh pi@mainsailos.local
pi@mainsailos.local's password: 
Linux mainsailos 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Jan  2 05:16:43 2021 from

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@mainsailos:~ $ $ dmesg -w -t
-bash: $: command not found
pi@mainsailos:~ $ dmesg -w -t
Booting Linux on physical CPU 0x0
Linux version 5.4.79-v7l+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 SMP Mon Nov 23 13:27:40 GMT 2020
CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
Memory policy: Data cache writealloc
Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
On node 0 totalpages: 999424
  DMA zone: 2304 pages used for memmap
  DMA zone: 0 pages reserved
  DMA zone: 196608 pages, LIFO batch:63
  HighMem zone: 802816 pages, LIFO batch:63
percpu: Embedded 20 pages/cpu s49804 r8192 d23924 u81920
pcpu-alloc: s49804 r8192 d23924 u81920 alloc=20*4096
pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Built 1 zonelists, mobility grouping on.  Total pages: 997120
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:59:DB:06 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=89d3ea0d-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
mem auto-init: stack:off, heap alloc:off, heap free:off
software IO TLB: mapped [mem 0x17cc0000-0x1bcc0000] (64MB)
Memory: 3602980K/3997696K available (10240K kernel code, 739K rwdata, 2816K rodata, 2048K init, 854K bss, 132572K reserved, 262144K cma-reserved, 3211264K highmem)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
ftrace: allocating 31111 entries in 61 pages
rcu: Hierarchical RCU implementation.
rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
GIC: Using split EOI/Deactivate mode
random: get_random_bytes called from start_kernel+0x344/0x518 with crng_init=0
sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
bcm2835: system timer (irq = 17)
arch_timer: cp15 timer(s) running at 54.00MHz (phys).
clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
Switching to timer-based delay loop, resolution 18ns
Console: colour dummy device 80x30
printk: console [tty1] enabled
Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
pid_max: default: 32768 minimum: 301
LSM: Security Framework initializing
Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Disabling memory control group subsystem
CPU: Testing write buffer coherency: ok
CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Setting up static identity map for 0x200000 - 0x20003c
rcu: Hierarchical SRCU implementation.
smp: Bringing up secondary CPUs ...
[... 631 lines cut ...]
Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
bcmgenet: Skipping UMAC reset
bcmgenet fd580000.ethernet: configuring instance for external RGMII
bcmgenet fd580000.ethernet eth0: Link is Down
IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Bluetooth: Core ver 2.22
NET: Registered protocol family 31
Bluetooth: HCI device and connection manager initialized
Bluetooth: HCI socket layer initialized
Bluetooth: L2CAP socket layer initialized
Bluetooth: SCO socket layer initialized
Bluetooth: HCI UART driver ver 2.3
Bluetooth: HCI UART protocol H4 registered
Bluetooth: HCI UART protocol Three-wire (H5) registered
Bluetooth: HCI UART protocol Broadcom registered
Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Bluetooth: BNEP filters: protocol multicast
Bluetooth: BNEP socket layer initialized
ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
can: controller area network core (rev 20170425 abi 9)
NET: Registered protocol family 29
can: raw protocol (rev 20170425)

Sineos, 2021-01-04:
Really weird behavior. I would guess there is little Klipper can do if the OS does not even enumerate the USB device. Sry, out of ideas now.

Roycinger, 2021-01-04:
Did you try powering the board by its Vin instead of usb only?

Arksine, 2021-01-05:
The presence of Maple and the `1eaf` vendor ID suggests that your board has the stm32duino bootloader flashed.  At one point there was a binary floating around that attempted to flash this bootloader in place using IAP (similar to how the `betterbootloader` works for lpc176x devices).  You state that you attempted to flash some binaries provided by others, is it possible that they gave you this one?  This was a very unstable process on the stm32f1 that didn't work out well which is why the binary was never officially released.

You also mentioned that your attempts to flash via sdcard have been successful (at least firmware.bin got renamed to firmware.cur), which suggests that you still have the stock bootloader.  Regardless, it sounds to me like you are experiencing a bootloader issue.  Perhaps the stock bootloader is writing flash correctly but not able to jump to the program.  To verify I would recommend flashing something else, perhaps Marlin, to see if it runs.

zellneralex, 2021-01-05:
Arksine the other was from me. I was not sure if there was any mistake by 
- setup
- Download
- Renaming
It was a purely klipper build and to my knowledge there was never tried anything else as a klipper build.
And yes the board was always connected to 24V during all tries.
He has now a new board and the flashing worked the first time.

iDestz, 2021-01-08:
As Alex said, we had also flashed the firmware from BTT. It's funny that he read out the serial directly after the first make. Just like last time, where it worked. After that I copied the serial and entered the printer.cfg. I did not move the printer or the cables. I restarted the Pi and it didn't read the serial anymore. As Alex said, I have now ordered a new board. Also a skr Mini e3 2.0. This had worked directly after flashing. So something must be broken on the board. Alex had described above where the problem could be.

With this I would also close the issue and thank you last to all who helped me to find out what it was and what you could do. Great community! Thanks a lot!

closed 2021-01-08 by iDestz