Klipper3d/klipper #3715: Erratic Filament Sensor

Sineos, 2021-01-02:
I just installed a [filament runout sensor](https://www.aliexpress.com/item/4000269547406.html?spm=a2g0s.9042311.0.0.4bc64c4d272cF6) and it is behaving erratically. Regardless of the pin configuration (pull-up, negated etc) it detects a run-out. Connected on my SKR Turbo 1.4 on the E0Det Pin.

Strange thing is that it also does in fact detect the filament when queried manually via `QUERY_FILAMENT_SENSOR SENSOR=filament_sensor`

```
Filament Sensor filament_sensor: filament detected
Filament Sensor filament_sensor: runout event detected, Time 1191148.91
Stats 1191148.9: gcodein=13101 mcu: mcu_awake=0.024 mcu_task_avg=0.000019 mcu_task_stddev=0.000024 bytes_write=93342 bytes_read=32433 bytes_retransmit=9 bytes_invalid=0 send_seq=2425 receive_seq=2425 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=53 stalled_bytes=4278 freq=119993887 rpi: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000016 bytes_write=1584 bytes_read=6560 bytes_retransmit=0 bytes_invalid=0 send_seq=247 receive_seq=247 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999517 adj=50002030 heater_bed: target=65 temp=65.1 pwm=0.102  print_time=1068.262 buffer_time=2.246 print_stall=0 extruder: target=215 temp=216.2 pwm=0.393 sysload=0.10 cputime=-1494.990 memavail=645300
Stats 1191150.0: gcodein=13140 mcu: mcu_awake=0.024 mcu_task_avg=0.000019 mcu_task_stddev=0.000024 bytes_write=98804 bytes_read=33001 bytes_retransmit=9 bytes_invalid=0 send_seq=2516 receive_seq=2516 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119993886 rpi: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000016 bytes_write=1590 bytes_read=6576 bytes_retransmit=0 bytes_invalid=0 send_seq=248 receive_seq=248 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999516 adj=50001984 heater_bed: target=65 temp=65.1 pwm=0.067  print_time=1068.367 buffer_time=1.350 print_stall=0 extruder: target=215 temp=216.1 pwm=0.393 sysload=0.10 cputime=-1494.940 memavail=645300
Stats 1191151.0: gcodein=13140 mcu: mcu_awake=0.024 mcu_task_avg=0.000019 mcu_task_stddev=0.000024 bytes_write=99245 bytes_read=33169 bytes_retransmit=9 bytes_invalid=0 send_seq=2528 receive_seq=2528 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119993882 rpi: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000016 bytes_write=1596 bytes_read=6592 bytes_retransmit=0 bytes_invalid=0 send_seq=249 receive_seq=249 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999518 adj=50001971 heater_bed: target=65 temp=65.1 pwm=0.000  print_time=1068.384 buffer_time=0.366 print_stall=0 extruder: target=215 temp=216.4 pwm=0.393 sysload=0.10 cputime=-1494.909 memavail=645300
Filament Sensor filament_sensor: filament detected
Filament Sensor filament_sensor: filament detected
Stats 1191152.0: gcodein=15681 mcu: mcu_awake=0.024 mcu_task_avg=0.000019 mcu_task_stddev=0.000024 bytes_write=107796 bytes_read=33996 bytes_retransmit=9 bytes_invalid=0 send_seq=2671 receive_seq=2671 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=9784 freq=119993885 rpi: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000016 bytes_write=1602 bytes_read=6608 bytes_retransmit=0 bytes_invalid=0 send_seq=250 receive_seq=250 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999520 adj=50002019 heater_bed: target=65 temp=65.1 pwm=0.030  print_time=1072.058 buffer_time=3.039 print_stall=0 extruder: target=215 temp=216.7 pwm=0.329 sysload=0.09 cputime=-1494.729 memavail=645300
Stats 1191153.0: gcodein=15681 mcu: mcu_awake=0.024 mcu_task_avg=0.000019 mcu_task_stddev=0.000024 bytes_write=113980 bytes_read=34624 bytes_retransmit=9 bytes_invalid=0 send_seq=2774 receive_seq=2774 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=7 stalled_bytes=4161 freq=119993891 rpi: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000016 bytes_write=1608 bytes_read=6624 bytes_retransmit=0 bytes_invalid=0 send_seq=251 receive_seq=251 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999517 adj=50002060 heater_bed: target=65 temp=65.0 pwm=0.186  print_time=1072.058 buffer_time=2.038 print_stall=0 extruder: target=215 temp=216.4 pwm=0.435 sysload=0.09 cputime=-1494.681 memavail=645300
Filament Sensor filament_sensor: filament not detected
Stats 1191154.0: gcodein=16657 mcu: mcu_awake=0.027 mcu_task_avg=0.000019 mcu_task_stddev=0.000024 bytes_write=120458 bytes_read=35278 bytes_retransmit=9 bytes_invalid=0 send_seq=2880 receive_seq=2880 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=38 stalled_bytes=5341 freq=119993890 rpi: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000013 bytes_write=1614 bytes_read=6653 bytes_retransmit=0 bytes_invalid=0 send_seq=252 receive_seq=252 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=49999518 adj=50002018 heater_bed: target=65 temp=65.0 pwm=0.186  print_time=1073.298 buffer_time=2.277 print_stall=0 extruder: target=215 temp=216.6 pwm=0.399 sysload=0.09 cputime=-1494.531 memavail=645300
Filament Sensor filament_sensor: filament detected
```

[image]

To debug I entered:
```
runout_gcode = 
	M117 Debug: Runout
insert_gcode = 
	M117 Debug: Insert
```
The `insert_gcode` seems to never fire, which is in bit of contradiction to the above log snippet.

In the end it may be a EMI issue or the thing is simply broken. Any idea what I can check / change would be appreciated. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5760064/klippy.log)

Sineos, 2021-01-02:
I disassembled the thing and I guess, this sensor type is simply not supported by Klipper:

[image]
[image]
[image]

This is a rotary encoder using a Schmitt-Trigger to generate pulses. 
Is there any chance to have this supported?

XSJFYZS, 2021-01-03:
https://plugins.octoprint.org/plugins/smartfilamentsensor/
Check this
But I really need klipper to support this Ir Filament Sensor, which is very cheap and can also detect stuck
https://www.aliexpress.com/item/32976742501.html

Sineos, 2021-01-04:
> https://plugins.octoprint.org/plugins/smartfilamentsensor/
> Check this

Thanks @XSJFYZS, already know it but I would really appreciate a direct integration.
I was hoping that @Arksine might be willing to spend some of his precious time as he is also the original author of the filament runout. I would also happily donate such a device if someone takes this up.

dmbutyugin, 2021-01-09:
If I understand correctly, that device has an encoder, which is very different from what the current filament runout sensor supports (the binary signal filament in/out).

Sineos, 2021-01-09:
> 
> 
> If I understand correctly, that device has an encoder, which is very different from what the current filament runout sensor supports (the binary signal filament in/out).

Yes exactly. It basically does not detect presence but movement. So it would need to be tied to the `G1 Exx` processing and could thus detect abnormal conditions: Run-out, jam etc.

If it would account for only positive extrusion it could even measure the consumed filament for a print. It though has only one output pin, so only hardware-wise it cannot distinguish between extrusion / retraction.

XSJFYZS, 2021-01-09:
> > 如果我理解正确，则该设备具有编码器，该编码器与当前灯丝跳动传感器支持的编码器（二进制信号灯丝输入/输出）有很大不同。
> 
> 对，就是这样。它基本上不检测存在，而是检测运动。因此，它将需要与`G1 Exx`处理相关联，从而可以检测到异常情况：跳动，卡纸等。
> 
> 如果仅考虑正向挤出，它甚至可以测量打印消耗的长丝。尽管它只有一个输出引脚，所以仅从硬件角度而言，它无法区分挤出/收缩。

Very strange, in the product description, it is an optical encoder

Sineos, 2021-01-09:
Well it is.
The little PCB has an IR array from which high / low impulses are generated via a Schmitt trigger. What looks like a toy wheel is the encoder disc.

Arksine, 2021-01-09:
FWIW, I don't believe it is enough to measure encoder position against incoming gcode commands, at least not if you want the sensor to be responsive.   Motion based sensors (whether its an encoder, laser, etc) need to be calibrated and measured against the real time extruder position.  As things stand today you have to query the MCU to get that information which isn't ideal.  Its possible that Klipper could add the ability to calculate the position without querying the MCU in the future though.   

Unfortunately I don't have time to work on this at the moment.   Hopefully someone else interested will be able to contribute.

Sineos, 2021-01-31:
#3857

PR #3857 (2021-03-15): Add filament_motion_sensor
Add functionality to support a Filament Motion Sensor for detecting extruder jams as well as runouts.  Works by an encoder toggling the switch_pin 0/1 as the filament is pulled through the sensor.

Everything seems to work ok, except I have seen some weirdness in pause_resume.  When pausing, the printer runs the runout_gcode however it sometimes moves back over the print job (does not affect Z, only X (and maybe Y)).  Increasing the pause_delay may resolve this.

Also pause_resume seemed a little inconsistent in how it was handled so I tried to address, not sure if t [... cut]
closed 2021-01-31 by Sineos