Klipper3d/klipper #3716: Hardware PWM for heaters on Duet

9ary, 2021-01-02:
This is more of a question I suppose.
I'd like to set my bed PWM frequency to 10kHz or more to get rid of the switching sound (it's a 24V MK52 bed clone on a Voron Switchwire).
Does Klipper use hardware PWM for the bed heater on this board? If not, is it detrimental for performance to set such a high PWM frequency? The only other issue I found about this was about a RAMPS board, so it's not very helpful.

9ary, 2021-01-02:
It looks like the noise I'm hearing comes from the bed itself and using such a high PWM frequency may not be a very good idea.
https://github.com/prusa3d/Prusa-Firmware/issues/1862#issuecomment-505302336
https://github.com/MarlinFirmware/Marlin/issues/18137

9ary, 2021-02-03:
I'm reopening this because I looked up the datasheets for the bed MOSFETs used by the Einsy Rambo (Prusa) and Duet 2 boards. The latter uses a much faster switching MOSFET which is intended for switch mode power supplies and should apparently dissipate an order of magnitude less heat because of switching losses (about 270mW for the Einsy, vs 45mW for the Duet, with a 10A bed at 32kHz), so it should be safe to drive at high frequencies.
I think it would also be interesting to implement something similar to Prusa's workaround in Klipper, but this probably deserves its own issue.
Sources:
- Einsy bed MOSFET: https://assets.nexperia.com/documents/data-sheet/PSMN1R8-40YLC.pdf
- Duet bed MOSFET: https://www.infineon.com/dgdl/Infineon-IPD036N04L-DS-v01_00-en.pdf?fileId=db3a3043163797a6011643476a6505a5
- Power losses in an SMPS: https://d1d2qsbl8m0m72.cloudfront.net/en/products/databook/applinote/ic/power/switching_regulator/power_loss_appli-e.pdf

9ary, 2021-02-25:
I still want this feature.

KevinOConnor, 2021-03-05:
If you want to increase the pwm frequency of the heater you can do that today with the `pwm_cycle_time` parameter.  (Though you almost certainly don't want to switch a mosfet 20K times per second.)

Given our updated policy on github tickets (see https://www.klipper3d.org/Contact.html ), I am closing this ticket.

-Kevin

closed 2021-03-05 by KevinOConnor

9ary, 2021-03-05:
> If you want to increase the pwm frequency of the heater you can do that today with the `pwm_cycle_time` parameter.

That's not hardware PWM though as far as I know, so it could cause performance issues, right?

> (Though you almost certainly don't want to switch a mosfet 20K times per second.)

Yes I do, switch-mode power supplies often operate north of 100kHz. Whether this is viable depends entirely on the transistor you pick, and as I've written in my previous comment, I believe the bed MOSFET on the Duet is suitable for this, unlike the one Prusa is using.

> Given our updated policy on github tickets (see https://www.klipper3d.org/Contact.html ), I am closing this ticket.

That's okay with me, I was planning on implementing this myself anyway, but I would have appreciated your input on the subject. Would you consider merging a PR for this?

KevinOConnor, 2021-03-05:
> That's not hardware PWM though as far as I know, so it could cause performance issues, right?

A SAM4E mcu easily do 20khz soft pwm.  As a "general rule of thumb", you can consider each soft pwm update to be roughly the same cost as a step.  So, you can look at the benchmarks to generally gauge capacity - https://www.klipper3d.org/Features.html#step-benchmarks

-Kevin

9ary, 2021-03-05:
Ah that's good to know, thanks!