Klipper3d/klipper #3717: GY-521 Resonances Support

wuast94, 2021-01-02:
Is there a way to support the GY-521 too? have them here and would safe me to buy new ones :)

Here is the datasheet i found: https://cdn.shopify.com/s/files/1/1509/1638/files/GY-521_6-Achsen_Gyroskop_und_Beschleunigungssensor_Datenblatt_AZ-Delivery_Vertriebs_GmbH.pdf?11553127782924451166

wuast94, 2021-01-02:
Here my lovly little script :* 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5760449/klippy.log)

and if thats the wrong place to ask that pls close this issue and referre me to the right place :) great work so far that you´ve all done here :)

dmbutyugin, 2021-01-04:
GY-521 only supports I2C, which does not have enough throughput to be useful for measuring resonances.

wuast94, 2021-01-04:
Ah OK thanks.. Than I will close this issue 😊

closed 2021-01-04 by wuast94