Klipper3d/klipper #3719: About menu.exit

yyxp1989, 2021-01-03:
hello，I have a problem with menu Settings
my display use mini12864

I want the function ：When I click on the menu [Load PLA Fill] and exit menu list return to the display main screen
I have read it https://www.klipper3d.org/Config_Reference.html#menu

how  to use main.exit() in that code

my code:
[menu __main __filament1]
type: list
name: Filament

[menu __main __filament1 __PLAloadf]
type: command
name: Load PLA Fill
gcode:
    SAVE_GCODE_STATE NAME=__filament__PLAloadf
    M117 PreHeating
    M104 S210
    M109 S210
    M117 Load Fill Now
    M83
    G92 E0.0
    G1 E70 F350
    G1 E80 F200
    G92 E0.0
    RESTORE_GCODE_STATE NAME=__filament__PLAloadf

Thank you very much if you can solve my problem！

mcmatrix, 2021-01-03:
Maybe something like this.
```ini
[menu __main __filament1]
type: list
name: Filament

[menu __main __filament1 __PLAloadf]
type: command
name: Load PLA Fill
gcode:
  SAVE_GCODE_STATE NAME=__filament__PLAloadf
  M117 PreHeating
  M104 S210
  M109 S210
  M117 Load Fill Now
  M83
  G92 E0.0
  G1 E70 F350
  G1 E80 F200
  G92 E0.0
  RESTORE_GCODE_STATE NAME=__filament__PLAloadf
  {menu.exit()}
```

yyxp1989, 2021-01-04:
> 也许是这样的。
> 
> ```ini
> [菜单__main __filament1]
> 类型：列表
> 名称：长丝[菜单__main __filament1 __PLAloadf]
> 类型：命令
> 名称：中国人民解放军负载填充
> 的G-码：
>   SAVE_GCODE_STATE NAME = __ filament__PLAloadf 
>   M117预热
>   M104 S210 
>   M109 S210 
>   M117装载填充现在
>   M83 
>   G92 E0.0 
>   G1 E70 F350 
>   G1 E80 F200 
>   G92 E0.0 
>   RESTORE_GCODE_STATE名称= __细丝
>   __PLAloadf {menu.exit（）}
> ```

Thanks， it works fine

closed 2021-02-03 by github-actions[bot]