Klipper3d/klipper #3720: TEST_RESONANCES = MCU 'mcu' shutdown: Timer too close

3Wings, 2021-01-03:
Klipper + Mainsail + Moonraker
Pi Zero W

ACCELEROMETER_QUERY
Gives me: // adxl345 values (x, y, z): -3451.940800, -2667.408800, 50759.220400

MEASURE_AXES_NOISE
Gives me: // Axes noise for xy-axis accelerometer: 45.509496 (x), 66.693895 (y), 49.452731 (z)

When typing: TEST_RESONANCES AXIS=X
it start but stops logging in mainsail after 31-34hz
but i can hear that it contonius with the test

Then i get this error in mainsail:

-----------------
MCU 'mcu' shutdown: Timer too close
This is generally indicative of an intermittent
communication failure between micro-controller and host.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
-----------------

If i ssh to pi to see if it have created logs for i x axis in /tmp/
i cant find any.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5762008/klippy.log)

3Wings, 2021-01-04:
Did read issue: 3650
Will test another sensor and shorter cables.
And reply here after new tests.

dmbutyugin, 2021-01-04:
The log is incomplete. From the logs and your description it seems that the main Klipper thread does not get enough CPU cycles or is blocked. It is either that the RPi is overloaded by some other processes running in parallel, or there are some issues in the Klipper setup (e.g. issues with Mainsail - Klipper integration?).

3Wings, 2021-01-04:
Can it be to much for a pi zero w, its single core 1ghz.

Can test on a pi 3b + tomorrow got some more power

3Wings, 2021-01-05:
Works now.

Same sensor and cables, but did change to a faster raspberry, did use it on a pi 3b+ today.

[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/5771520/klippy.log)

closed 2021-01-06 by 3Wings