Klipper3d/klipper #3721: Klipper host process crash at RESTART

gkatev, 2021-01-03:
Hi,
Lately the klipper process/service on the Pi (Orange Pi Zero) has been crashing with SIGABRT, right after issuing a RESTART command (once every 2-3 restarts).

From the journal:
```
Dec 29 11:05:57 3dpcl python[26241]: Fatal Python error: Invalid thread state for this thread
Dec 29 11:05:58 3dpcl systemd[1]: klipper.service: Main process exited, code=killed, status=6/ABRT
Dec 29 11:05:58 3dpcl systemd[1]: klipper.service: Failed with result 'signal'.
Dec 29 11:06:08 3dpcl systemd[1]: klipper.service: Service RestartSec=10s expired, scheduling restart.
Dec 29 11:06:08 3dpcl systemd[1]: klipper.service: Scheduled restart job, restart counter is at 1.
Dec 29 11:06:08 3dpcl systemd[1]: Stopped Klipper daemon.
Dec 29 11:06:08 3dpcl systemd[1]: Started Klipper daemon.
```
Backtrace from the coredump:
```
PID: 26241 (python)
           UID: 1000 (user)
           GID: 1000 (user)
        Signal: 6 (ABRT)
     Timestamp: Tue 2020-12-29 11:05:57 EET (2min 11s ago)
  Command Line: /home/user/klippy-env/bin/python /home/user/klipper/klippy/klippy.py /home/user/printer.cfg -l /tmp/klippy.log
    Executable: /home/user/klippy-env/bin/python2.7-dbg
 Control Group: /system.slice/klipper.service
          Unit: klipper.service
         Slice: system.slice
       Boot ID: 521322b0c1f844cd9e372f5e0ed034ac
    Machine ID: 05c0c8227a0448908770f93a976a3985
      Hostname: 3dpcl
       Storage: /var/lib/systemd/coredump/core.python.1000.521322b0c1f844cd9e372f5e0ed034ac.26241.1609232757000000.lz4
       Message: Process 26241 (python) of user 1000 dumped core.
                
                Stack trace of thread 27587:
                #0  0x00000000b6d2f746 n/a (libc.so.6)

GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/user/klippy-env/bin/python2.7-dbg...done.

warning: core file may not match specified executable file.
[New LWP 27587]
[New LWP 26241]
[New LWP 26243]
[New LWP 27586]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `/home/user/klippy-env/bin/python /home/user/klipper/klippy/klippy.py /home/user'.
Program terminated with signal SIGABRT, Aborted.
#0  0xb6d2f746 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
[Current thread is 1 (Thread 0xb5477460 (LWP 27587))]
(gdb) bt
#0  0xb6d2f746 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
#1  0xb6d3d0ae in raise () from /lib/arm-linux-gnueabihf/libc.so.6
#2  0xb6d2f1f2 in abort () from /lib/arm-linux-gnueabihf/libc.so.6
#3  0x005454aa in Py_FatalError (msg=0x619f24 "Invalid thread state for this thread") at ../Python/pythonrun.c:1712
#4  0x00541cf4 in PyThreadState_Swap (newts=0x167ed60) at ../Python/pystate.c:354
#5  0x00512c1e in PyEval_AcquireThread (tstate=0x167ed60) at ../Python/ceval.c:278
#6  0x005f061c in t_bootstrap (boot_raw=0x150d768) at ../Modules/threadmodule.c:618
#7  0x00555dfa in pythread_wrapper (arg=0xb4100f58) at ../Python/thread_pthread.h:178
#8  0xb6edcbbe in start_thread () from /lib/arm-linux-gnueabihf/libpthread.so.0
#9  0xb6db262c in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```
I'm not even sure to what extent this is related to or caused by klipper, and if the coredump helps in any way, but any insight is appreciated.

I haven't really changed anything on my Orange Pi (problem first appeared 2-3 weeks ago), apart from updating Klipper, Octoprint, and the system. This could (?) even have something to do with Armbian's libraries, but since it does consistently trigger when I issue the RESTART command.. what might it be?

[klipper.log](https://github.com/KevinOConnor/klipper/files/5762055/klipper.log)
[core.python.1000.521322b0c1f844cd9e372f5e0ed034ac.26241.1609232757000000.zip](https://github.com/KevinOConnor/klipper/files/5762058/core.python.1000.521322b0c1f844cd9e372f5e0ed034ac.26241.1609232757000000.zip)

KevinOConnor, 2021-01-24:
The typical reason these types of errors occur is a corruption of the sd card that the Linux operating system is running on.  It looks like you're using an "orange pi" - in that case, it may also be due to something incompatible in the system dependencies.  Unfortunately, we don't support the "orange pi", so there's not much we can offer for advice.

-Kevin

gkatev, 2021-01-25:
This is a bizarre issue indeed. I'll look into the SD card's health, but this is the only problem that I've encountered, always triggered at RESTART. AFAIK OrangePi & Armbian are pretty stable, and they are interesting devices (not asking to support them :) ).

I can snoop around the code and try to find what's causing this; it's quite reproducible. Any ideas to direct towards where the problem might reside? Or anything about the restart procedure that's maybe tricky or not too obvious from the code alone?

Thanks

gkatev, 2021-02-17:
So, I'm not 100% sure but I think after I switched to using FIRMWARE_RESTART instead of RESTART the problem has not re-occurred. Does that make some sort of sense or hint towards something? I though the FIRMWARE_RESTART command was a super-set of RESTART (?).

gkatev, 2021-03-12:
Don't know if there's a point keeping it open, but it is still an issue, so let's keep it open for now...

closed 2021-04-10 by github-actions[bot]