Klipper3d/klipper #3722: Caribou/LDO OLED Display shows garbage. (Sometimes)

nofuturekid, 2021-01-03:
Hi,
I have an Caribou 320 with an Einsy Board and a LDO OLED display. It's configured as **hd44780**. Once every 5-6 firmware restarts it's working correctly but in most of times it shows just garbage characters or the rows/columns are somehow shifted.

The display itself is not broken as it's working fine with the Caribou firmware. The Caribou team also had some trouble with these and needed to adjust the Prusa firmware to get the LDO OLED working with it.

Maybe someone can give it a try as my coding skills are very limited.

nofuturekid, 2021-01-03:
Adding log file.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5762361/klippy.log)

KevinOConnor, 2021-01-24:
It doesn't look like you're running code from this repo, so there's not much we can do to help.  You should check that the code you are running has the changes from October 2020 (in particular commit 65a49d17).

-Kevin

nofuturekid, 2021-01-24:
What makes you think I'm not running from this repo? Ohh yes, because the "s-curve" thingy...
But it doesn't matter - it's the same with code from your repo.

But never mind, just yesterday found a workaround:
/klippy/extras/display/hd44780.py
replacing: "init = [[0x33], [0x33], [0x32], [0x28, 0x28, 0x02]]" with "init = [[0x0]]" solved it reliably for me.

nofuturekid, 2021-01-24:
Tried with both init lines:
"init = [[0x33], [0x33], [0x33, 0x22, 0x28, 0x02]]" - Display never show data correct, no matter how ofter resetting
"init = [[0x33], [0x33], [0x32], [0x28, 0x28, 0x02]]" - Display works maybe every 6th or 7th time after restart

"init = [[0x0]]" - Display works everytime

nofuturekid, 2021-01-24:
Still not sure how to put that into code which then could be added permantly into your repo. So that I don't have to patch this every single time after a git pull.

KevinOConnor, 2021-01-24:
I'm not sure either.  It looks like the display on that printer doesn't quite follow the hd44780 protocol.  We can't remove the 8bit/4bit init for real hd44780 displays and it seems your display doesn't like it.

BTW, I think you'd want: `init = [[0x02]]`

-Kevin

nofuturekid, 2021-01-24:
I have two displays. One "normal" LCD on my Prusa which works fine with the init for the hd44780 and on my Caribou it's that bitchy LDO OLED.

As said the guys over at Caribou had to patch their Firmware (PrusaFW fork) to use these kind of display also.

I will try your suggestion with `init = [[0x02]]`.
What about extending the `[display]` with an optional parameter which then changes the initialisation string?

nofuturekid, 2021-01-24:
Some like
`[display]
lcd_type = hd44780
lcd_oled = true  <- new optional parameter to change init string to [[0x02]]
rs_pin = PD5
e_pin = PF7
d4_pin = PF5
d5_pin = PG4
d6_pin = PH7
d7_pin = PG3
encoder_pins = ^PJ1,^PJ2
click_pin = ^!PH6
`

KevinOConnor, 2021-01-24:
That's fine if you or someone else wishes to implement and submit it.  FWIW, I'd probably go with something like `hd44780_protocol_init: False`

-Kevin

nofuturekid, 2021-01-24:
I will see what I'm able to achieve. As said I'm not a coder.

nofuturekid, 2021-01-24:
I hope you can accept this pull request.

PR #3838 (2021-02-01): hd44780: Add hd44780 LDO OLED protocol init
The LDO OLED needs a different init.
Fixes: #3722
commit 60e4cddf (2021-02-01): hd44780: Add hd44780 protocol init config
LDO OLED needs a different init string.

Fixes: #3722
closed 2021-02-01 by KevinOConnor