Klipper3d/klipper #3723: FR add enter_gcode and exit_gcode to [display]

zellneralex, 2021-01-03:
Here is the problem as I do not use the display that much I dim it down via a delay_gcode at klipper start and have two other gcodes to make it bright and dim it. At the moment I use the web_ui to activate this gcodes. But it would much better if I could add both to the display section and the 
enter_gcode: Would be execute as you enter the menu
exit_gcode: Would be executed as you leave the menu

As I workaround I might end up to solder a wire from the click_pin to an other one to do it via a gcode_button

mcmatrix, 2021-01-19:
Hi @zellneralex 

I'll check what options we have with current menu implementation and what can be added as development.

mcmatrix, 2021-01-19:
@KevinOConnor I'm asking your opinion on this.
The menu is triggering an event on beginning and exit.

I was thinking maybe it useful to implement a new generic module
for running g-code on a specific event trigger.
Example:
```
[gcode_event my_gcode_event_listener]
event:
#   The event name we are listening. This parameter must be provided.
#gcode:
#   A list of G-Code commands to execute when the event is triggered.
#   G-Code templates are supported. This parameter must be provided.
```

or just adding 2 new g-code parameters to display and execute these on exit and start?

zellneralex, 2021-01-19:
Thank you for looking in to it.
Maybe the easiest might be to add this to main menu entry point. then you would not need to add it to the current [display] section. 
```### menu main ###
[menu __main]
type: list
name: Main

KevinOConnor, 2021-01-19:
> I was thinking maybe it useful to implement a new generic module
for running g-code on a specific event trigger.

I don't think that will work well.  The internal events have state associated with them and can have specific timing requirements (eg, that the called function not pause).  I don't think it is practical to run arbitrary gcode on an arbitrary internal event.

FWIW, I fear running custom g-code on menu enter/exit will have higher costs than benefit.  FWIW, this is an area that I think may be best served with a custom python module (perhaps not even distributed with the master klipper branch), or alternatively by accepting that these displays just aren't pretty and not attempt to implement a solution.  YMMV.

-Kevin

closed 2021-02-18 by github-actions[bot]