Klipper3d/klipper #3726: Bowden Extruder, Pressure Advance Bulge After 90 Degree Cornering Move

Nyshan, 2021-01-04:
I've been running back and forth on this issue for close to two weeks now with very little improvement seen so far. My problem is  basically an exact copy of the behaviour detailed in this issue: https://github.com/KevinOConnor/klipper/issues/1260#issuecomment-464094072 and I've found that using the listed "solution" here: https://github.com/KevinOConnor/klipper/issues/1260#issuecomment-465647421 seems to provide better cornering but there are still bulges, albeit somewhat less pronounced.

I can provide more pictures in a few hours to better show the specific problems I'm seeing.

extruder: Mobius 4 (TMC2209@1.1A, MT-1704HSM168RE)
default max XYE acceleration: 4000
default max velocity: 200
default square corner velocity: 5
default PA value: Tested values between 0.50-0.70 with same results until corners begin to rapidly deteriorate after 0.7

Printing speed: 120mms (minimum layer time and volumetric flow rate limit disabled)

Please let me know what else I can provide in regards to this issue (I will provide pictures of Square Corner test with PA 0.0-1.0 as well as samples of cubes printed between PA 0.5-0.7 once I get off work.)

Nyshan, 2021-01-04:
[image]

[image]

[image]

[image]

Sineos, 2021-01-04:
Are you using a 0.4 nozzle?

Nyshan, 2021-01-04:
> Are you using a 0.4 nozzle?

Yes, I am. This shouldn't matter but I'll include it as well:

Retract Speed: 40
Retract Distance: 4mm (I've run values up and down between 2-4)
Unretract Speed: 40
Material: ABS
Hotend: 250C

Sineos, 2021-01-04:
Well, with a 0.4 nozzle and layer-height <= 0.2 and 120 mm/s and accel 4000 you will never get reliable extrusion results. You are hitting the physical limits of the amount of molten plastics that can be pushed through a 0.4 opening.

Nyshan, 2021-01-04:
That's only 9.6mm^3/s. The recommended volumetric flowrate for testing PA in klipper on an 0.4 nozzle (assuming 0.4 extrusion width, 0.3 layer height, 100mm/s) is around 12mm^3/s so I have trouble understanding why this would be an issue.

Sineos, 2021-01-04:
Maybe for PLA you get away with this when increasing the nozzle temperature. But for ABS with 120 mm/s we are already reaching 14.4 mm^3/s (0.3 / 0.4 / 120).
At least to my experience, filaments like ABS or ASA will start giving problems at around 8 to 10 mm^3/s. 

YMMV

Nyshan, 2021-01-04:
So if I slow down to 80-100mm/s you'll acknowledge the problem I'm reporting?

Sineos, 2021-01-05:
To acknowledge that Klipper's PA has a general issue, just because you are having issues, would be quite pretentious, don't you think?

My theory is quite simple:
* Your acceleration is very high
* Your extrusions settings are at the upper limit (possibly exceeding it) of what is physically possible
* Your extruder is very powerful but also has a very high gear ratio (4:1?). This makes it difficult to quickly react on very fast changes
* You are using a bowden, which introduces additional losses

So there is more pressure build up in the nozzle than the PA algorithm and/or the physical properties of your system can compensate in a very short time. This then leads to over-extrusion at the start of a new acceleration phase.

Nyshan, 2021-01-05:
> To acknowledge that Klipper's PA has a general issue, just because you are having issues, would be quite pretentious, don't you think?
>
>
>
> My theory is quite simple:
>
> * Your acceleration is very [...]

Sorry if I came off as a bit abrasive, that wasn't my intention, I've done a significant amount of testing with acceleration and speed that leads me to believe that it isn't related to my issue. With that said all of that testing has been disjointed and undocumented. My next few tests can involve speed and acceleration. Would reducing speeds and accels down to 80mm/s and 2000mm/s respectively be enough or would you prefer to see a further initial reduction?

Additionally I don't believe that PA has a "problem" I think that something (other than just volumetric flowrate alone) in my config/slicer/printer is probably causing the issue but I have yet to make any real progress with config/slicer settings.

Sineos, 2021-01-05:
If you are fighting for every minute of print time, then I guess you will have to find your limits by iterating to the maximum possible.
I do not know your printer and I'm not a 3D printing expert by any means but maybe you find something useful:
* IMO an acceleration of 4000 is quite extreme and requires real sturdy and finely tuned mechanics (3000 is more or less a "default" for our consumer grade machines and 4000 equals +33%)
* Do you have set `max_accel_to_decel`? Your default without a setting would be `0,5 * max_accel`. If you are higher with this try to be more conservative 
* Try playing with `square_corner_velocity` to give the system a little break on sharp corners, maybe 4 to see if something changes
* https://teachingtechyt.github.io/calibration.html#accel is a nice guide for finding your max feedrate 
* The genuine Capricorn XS bowden tubes do make a difference 
* For speeds over 110 mm/s, I'm using a 0.6 nozzle because my (bowden / BMG Bondtech) extruder starts acting up otherwise

Good luck and if you find anything useful a feedback would be appreciated.

Sineos, 2021-01-05:
Hmm, on a second thought....you are sure to issue
`SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=1 ACCEL=500`
before starting the PA tuning? This should already set quite conservative limits.

Nyshan, 2021-01-05:
Yes, I'm issuing that command along with the tuning tower command. I can
see differing results along the corners of the tower but I never really
note a corner without a bulge before the corners begin to look like
fillets.

Sineos, 2021-01-06:
In my experience the range between bulges and fillets is quite small depending on the type of filament.
* PA seems the most forgiving. On my printer PA around 0.5
* PETG is already more critical. PA around 0.7
* ASA (my mostly used filament, I don't like ABS) is even worse. PA around 0.8

For reference, this is a PETG cube I printed yesterday:

- 80 mm/s
- 0.3 mm layer height
- 0.5 mm layer width
- Judging by my ears the extruder was already grinding occasionally, so definitively not a setting I would choose for printing

[image]

Nyshan, 2021-01-06:
Are you running a 0.6mm nozzle?

Sineos, 2021-01-06:
No, this currently is a 0.4 mm nozzle at 240 °C, but as I said, the extruder was already occasionally grinding (12 mm^3/s). For PETG or ASA I usually try to stay well below 10 mm^3/s.

Nyshan, 2021-01-07:
I'm almost wondering if I have a loose grub screw somewhere as I can't seem to achieve any improvement even at 0.85 PA on a bowden less than 20cm. I'm going to teardown my extruder just in case.

Nyshan, 2021-01-08:
Mechanical issues with extruder build can be ruled out at this point.

Sineos, 2021-01-08:
I'm out of ideas. Whereas the max flowrate IMO more depends on the nozzle diameter than on the melting capabilities of the hotend. I'm running a Mosquito Magnum myself and you can only push a certain amount through a 0.4 hole.

Have you tried if the effect is the same with a less demanding filament like PLA?

Nyshan, 2021-01-08:
> I'm out of ideas. Whereas the max flowrate IMO more depends on the nozzle diameter than on the melting capabilities of the hotend. I'm running a Mosquito Magnum myself and you can only push a [...]

Slowing down perimeter speeds to 60 inner, 30 outer shows a drastic improvement in PA performance; corners actually start to develop noticeable edges. So, as you previously implied, maybe I'm printing too damn fast... I am interested to see if the same behavior I've seen in my bowden at high speeds is also demonstrated in a geared direct drive system.

closed 2021-01-11 by klipper-gitissuebot