Klipper3d/klipper #3727: Can't find Step_distance for my ender 3 pro

thelandinlook, 2021-01-05:
I've just installed Klippy (quite new to 3d printing) and I'm reading everywhere that I need to find my step_distance to calibrate extrusion. I just can't find it anywhere and it's driving me mad. Youtube videos show it in the config file, but nothing. I scoured the logs, configs, github, nothing. What am I missing? 

Thanks. x  
[klippy.txt](https://github.com/KevinOConnor/klipper/files/5768626/klippy.txt)

zellneralex, 2021-01-05:
step_distance get replaced by rotation_distance. The filling might help https://www.klipper3d.org/Rotation_Distance.html#extruder

thelandinlook, 2021-01-06:
> step_distance get replaced by rotation_distance. The might help https://www.klipper3d.org/Rotation_Distance.html#extruder

Brilliant, sorted everything. thanks :)

closed 2021-01-06 by thelandinlook