Klipper3d/klipper #3728: SKR mini E3 V1.2 - Timer too close / lost mcu conenction

Stephan3, 2021-01-05:
Hello.

For testing i installed two E3 V1.2 i had arround to one if my Vorons. It feels like random that it throws timer too close / lost mcu conenctions midprint. I had 24h prints running fine, just repeat the same plate which failed after 4? hours.

One time it told timer too close, the other time it told me lost communication. i kept both logfiles.
https://www.file-upload.net/download-14420671/klippy_timertoclose.log.html
https://www.file-upload.net/download-14420672/klippy_communication_lost.log.html

The host system has not changed for 3 Years now. It a Atom X5 minipc. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/5769739/klippy.log)

Thanks
Stephan

KevinOConnor, 2021-01-05:
You'll need to attach the logs here if you wish to use this github issue tracker.  The docs explain how to compress the logs if they are large.

-Kevin

Stephan3, 2021-01-05:
EDIT - Post updated

ankle-hunter, 2021-01-09:
i can't get klipper to work on the skr mini e3 v1.2. i tried every combination. my printer just won't boot. and yeah i changed the klipper.bin file to firmware.bin file and put it on a sd card and in the printer and start the printer. can you maybe upload your klipper.bin file?

Stephan3, 2021-01-11:
@ankle-hunter 
i just did it as its stated in the config example header. But here it is attached.
[klipper.bin.zip](https://github.com/KevinOConnor/klipper/files/5794896/klipper.bin.zip)

Stephan3, 2021-01-11:
I replaced one of the Boards on 08th and it now seems to work. i close this for now.

closed 2021-01-11 by Stephan3

ankle-hunter, 2021-01-11:
> I replaced one of the Boards on 08th and it now seems to work. i close this for now.

Thanks!

ankle-hunter, 2021-01-11:
> I replaced one of the Boards on 08th and it now seems to work. i close this for now.

even with your .bin file the printer won't start. the screen of the ender 3 pro is just blue. nothings happens. klipper keeps saying 

Printer is not ready
The klippy host software is attempting to connect. Please retry in a few moments.
Klipper state: Not ready
Printer is not ready