Klipper3d/klipper #3729: [HELP!!!] BTT GTR v1.0 and Raspberry Pi 3 B + do not connect

DEAth1981, 2021-01-05:
Good day!

Available:
- BTT GTR v1.0;
- Raspberry Pi 3 B +.

Did:
1. Connected pins (GPIO14, GPIO15, GND) on Raspberry Pi 3 B + with pins (PA9, PA10, GND) on BTT GTR v1.0.
2. Installed B + OctoPrint and Klipper on Raspberry Pi 3.
3. Compiled the firmware for BTT GTR v1.0 according to the screenshot:
[image]
4. Copied "klipper.bin" to MicroSD BTT GTR v1.0, before renaming the file to "FIRMWARE.CUR".
5. Created "printer.cfg" using "generic-bigtreetech-gtr.cfg" as an example
6. In Raspberry Pi 3 B + disabled Linux serial console and got free "/dev/ttyS0".
7. In the connection settings OctoPrint set the serial port "/tmp/printer" and the baud rate "250000".
8. Rebooted BTT GTR v1.0 and Raspberry Pi 3 B +.

What's happening:
1. In OctoPrint, I click the "Connect" button and get the "Klipper: Connected to host" status.
2. I am trying to do something on the printer, for example, shuffle along the X axis.
3. I get errors with different frequency:
- "Can not update MCU 'mcu' config as it is shutdown";
- "Must home axis first: 10.000 0.000 0.000 [0.000]";
- "Shutdown due to M112 command".
4. After any error appears, I get the status "Klipper: Disconnected from host".

What am I doing wrong? How do you achieve stable operation of this bundle on Klipper?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5772377/klippy.log)

Hywelmartin, 2021-01-06:
FIRMWARE.BIN

DEAth1981, 2021-01-06:
> FIRMWARE.BIN

I will probably surprise you, but BTT GTR v1.0 absolutely does not care what extension the file has (.BIN or .CUR). Moreover, if you copy the file "FIRMWARE.BIN" to MicroSD and paste it into BTT GTR v1.0, then after turning it on, the file is automatically renamed to "FIRMWARE.CUR".
Just in case, I once again copied the firmware to MicroSD, but now I renamed the file to "FIRMWARE.BIN" ... NOTHING HAS CHANGED, i.e. my problem is NOT SOLVED!

Hywelmartin, 2021-01-06:
The controller changes the extension after flash..  so it "cares"..

DEAth1981, 2021-01-06:
> The controller changes the extension after flash.. so it "cares"..

As you say, but the essence of the matter has not changed... as the problem was, it remains

DEAth1981, 2021-01-06:
> Good day!
> 
> Available:
> 
> * BTT GTR v1.0;
> * Raspberry Pi 3 B+.
> 
> Did:
> 
> 1. Connected pins (GPIO14, GPIO15, GND) on Raspberry Pi 3 B+ with pins (PA9, PA10, GND) on BTT GTR v1.0.
> 2. Installed OctoPrint and Klipper on Raspberry Pi 3 B+.
> 3. Compiled the firmware for BTT GTR v1.0 according to the screenshot:
>    [image]
> 4. Copied "klipper.bin" to MicroSD BTT GTR v1.0, before renaming the file to "FIRMWARE.CUR".
> 5. Created "printer.cfg" using "generic-bigtreetech-gtr.cfg" as an example
> 6. In Raspberry Pi 3 B+ disabled Linux serial console and got free "/dev/ttyS0".
> 7. In the connection settings OctoPrint set the serial port "/tmp/printer" and the baud rate "250000".
> 8. Rebooted BTT GTR v1.0 and Raspberry Pi 3 B+.
> 
> What's happening:
> 
> 1. In OctoPrint, I click the "Connect" button and get the "Klipper: Connected to host" status.
> 2. I am trying to do something on the printer, for example, shuffle along the X axis.
> 3. I get errors with different frequency:
> 
> * "Can not update MCU 'mcu' config as it is shutdown";
> * "Must home axis first: 10.000 0.000 0.000 [0.000]";
> * "Shutdown due to M112 command".
> 
> 1. After any error appears, I get the status "Klipper: Disconnected from host".
> 
> What am I doing wrong? How do you achieve stable operation of this bundle on Klipper?
> [klippy.log](https://github.com/KevinOConnor/klipper/files/5772377/klippy.log)

I was already so worried that I decided to disconnect communication via GPIO and connected via USB, i.e. Pi -> USB cable -> GTR.
Recompiled firmware with USB enabled.
I fixed the configuration to use the new port, but nothing changed - ERRORS REMAINED!
I'm starting to hate this Klipper and also this GTR board !!!

Arksine, 2021-01-07:
You can't move prior to homing with Klipper, it will send an error back to the console.  It has been a while since I have used octoprint, but IIRC its default configuration might send M112 to the printer when it receives an error via the console.

I would recommend executing a FIRMWARE_RESTART.  When Klipper reports ready, home the axes first, then try to move.  If that works, search the Octoprint configuration for whatever option sends M112 on an error and disable it.

DEAth1981, 2021-01-07:
So I could not get this "set" to connect via UART. I ended up connecting the board and Pi via a USB host on the board. Everything works now.
I think that the problem was in the Raspberry Pi 3 B+, or rather in the peculiarities of its two UART ports... it was necessary to buy a Raspberry Pi 4 B.

closed 2021-01-07 by DEAth1981