Klipper3d/klipper #3730: Unknown value 'P1.17' in enumeration 'pin'

TheBigF1sh, 2021-01-06:
Help needed.

I have previously had Klipper working with the same hardware but recently had to start again with the Rasp Pi and OctoPi. Since then I have been unable to use Klipper - I largely reused my previous Printer.cfg 

Any help appreciated 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5776189/klippy.log)

Recv: // Unknown value 'P1.17' in enumeration 'pin'
Send: N0 M110 N0*125
Recv: // This type of error is frequently caused by running an older
Recv: // version of the firmware on the micro-controller (fix by
Recv: // recompiling and flashing the firmware).
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Changing monitoring state from "Detecting serial connection" to "Operational"
Recv: // Protocol error connecting to printer

jakep82, 2021-01-06:
The error is pretty descriptive.  You need to compile and flash new firmware on the board.

TheBigF1sh, 2021-01-06:
I did that - for the SKR Mini 1.2 this is completed by moving the firmware.bin to the sdcard and not using the MAKE FLASH cmd. But still nothing .....

jakep82, 2021-01-06:
If you're talking about the SKR Mini E3 v1.2, then your entire printer.cfg is wrong.  See the sample for proper pins.

https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v1.2.cfg

TheBigF1sh, 2021-01-06:
Let me recheck what I have done here and revert - and thanks for the swift response

TheBigF1sh, 2021-01-06:
Using that config verbatim with the exception of the MCU there is now an issue with reading the serial port (I am a little perplexed how the configuration in my previous printer config is misaligned)

Attached is the log

[klippy.log](https://github.com/KevinOConnor/klipper/files/5777107/klippy.log)

When typing "ls /dev/serial/by-id/*" via SSH i now get 

ls: cannot access '/dev/serial/by-id/*': No such file or directory

vs previously (with incorrect printer.cfg I got /dev/serial/by-id/usb-Klipper_stm32f103xe_37FFD6054242353921632257-if00

jakep82, 2021-01-06:
Try `lsusb` to see if the board is being recognized when it's connected via USB.  If it's not recognized by the Pi, then I'm not sure where to go from there.

TheBigF1sh, 2021-01-06:
pi@octopi:~/klipper $ lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

jakep82, 2021-01-06:
It doesn't see the board at all.  All I can suggest is reflashing it, and reconnect to the Pi after flashing.

TheBigF1sh, 2021-01-06:
Retried but to no avail (same result from lsusb as above)

TheBigF1sh, 2021-01-06:
reinstalled octopi and reflashed and now working ?!

closed 2021-01-06 by TheBigF1sh