Klipper3d/klipper #3732: Consistently bad calibration cubes.

thelandinlook, 2021-01-06:
After quite a lot of tuning and good results using the calibration files included in the klipper docs, I'm still consistently getting these lines on my calibration cube which I wasn't before Klipper. I've tried lowering the velocity, accelleration, corner velocity, temperature, and turning off pressure advance. I've also tried a much higher quality MK8 Düse Plated Copper Nozel.
Any ideas?

[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/5778657/klippy.log)

Thanks for your time. x

thijspomper, 2021-01-07:
Could you try it again with the default ender profile in Cura? So that we can eliminate that factor.

thelandinlook, 2021-01-07:
Resetting it did make it considerably better, I'll try and keep refining the calibration, a lot I still don't completely understand. Would you be able to tell me how much pressure advance replaces retraction amount? Does printing faster with pressure advance mean I should use higher retraction or lower? Does retraction speed alter that?

Thank you x

grigi, 2021-01-11:
That looks like uneven layer heights to me. It's likely caused by backlash.
If you have lead-screws on the Z, they have backlash, which is slight movement.

I experienced three different issues related to Z backlash:, with three different solutions:
1. Disabling z-hop (or force-enabling depending on bed movement) could have a significant impact. Basically you want to ensure that the movement is always from the bottom up, so that the bed can rest on the lead-screws.
2. Too much grease on the lead-screws actually making the lead-screw uneven. Clean thoroughly and add a thin layer of lubricant.
3. If you have a direct-drive extruder, the filament (which often runs over the top) can lift the gantry when either a sideways movement or an extrusion happens. You would want neither. Add a reverse bowden tube that is mounted on your toolhead, and mounted on the frame, so that gantry movement doesn't change the length of the filament, and now there is no way for the filament to tug your gantry up.

thelandinlook, 2021-01-19:
I returned that particular printeer

closed 2021-01-19 by thelandinlook