Klipper3d/klipper #3733: FR/question -- support for  I2C multiplexers

jtrmal, 2021-01-07:
I wonder if you think adding support for multiplexers would be of interest. For example, when a user would be interested in having a multitude of temp sensors on each stepper and who knows where else. It's oftentimes not easy to change the I2C address (if possible at all). I wasn't thinking about it too much but the config might be quite simple, in the style
```
[i2c_multiplex mplex]
i2c_address:
i2c_bus:
i2c_speed:
...

[temperature_sensor my_sensor]
sensor_type: BME280
i2c_address: mplex:1:177 
```
or something like that (for the address there would have to be inserted the multiplexing id and the sensor id -- but could perhaps be done some other way)
I can do C or C++ and have some experience with these things (in arduino), so I could even work on it (also as a way how to thank to Voron community and to Klipper community) if you think it would be a worthy addition and perhaps would be able to navigate me a bit what and where has to be changed.

KevinOConnor, 2021-01-10:
FYI, there is already some support for this (for example, see klippy/extras/sx1509.py).  However, it's worth pointing out that an alternative is to deploy a low-cost micro-controller that can interact directly with the klipper host software.  Often that's less expensive, simpler to deploy, and does not require any code changes.

-Kevin

jtrmal, 2021-02-01:
@KevinOConnor thanks for the response! And sorry for getting this late to it.

closed 2021-02-01 by jtrmal