Klipper3d/klipper #3734: Klipper cannot connect to printer after adding [mcu host] - Fluidd

oeijoenj, 2021-01-07:
Currently using Fluidd for klipper and want to attach temp sensor for enclosure and filament dryer.
I added raspberry as secondary MCU using this instructions : https://www.klipper3d.org/RPi_microcontroller.html?fbclid=IwAR2iiiCnqNr5wsTN94ba15-pKVp2va9qWwv7TDPCCDOvwzOkqxM3Pb47QAM

After i addded [mcu host] in printer.cfg and dod firmaware_restart. It gave error :
Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

Could you help ?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5779518/klippy.log)
[moonraker.log](https://github.com/KevinOConnor/klipper/files/5779519/moonraker.log)

3Wings, 2021-01-07:
From log File:
Unable to open port: [Errno 2] No such file or directory: '/tmp/klipper_host_mcu'

oeijoenj, 2021-01-07:
Is that due to I had error when execute make flash ?

Please specify FLASH_DEVICE

make: *** [src/stm32/Makefile:68: flash] Error 255.

Attached is my menuconfig, I'm using creality ender 3 v2.
[image]

FHeilmann, 2021-01-07:
your menuconfig is for your microcontroller board. In addition to your microcontroller board flash, you need to rerun menuconfig, select the "Linux process" option, make and make flash.

This is all outlined in the instructions, I suggest you try going through them again and focus on the details.

oeijoenj, 2021-01-07:
Got it. Sorry, I missed that statement "start by configuring.."
Thanks for your assistance. It's working now

closed 2021-01-07 by oeijoenj