Klipper3d/klipper #3735: Installed inductive sensor, led indicates but bed keeps moving.

cyclonesurfer, 2021-01-07:
I installed an inductive sensor on my skr 1.4 and when I home the printer the led indicates but the printer continues to move the z axis.

The sensor is a normal 6-36v inductive sensor with three wires, vsg.

I believe I have it configured correctly, so i'm stumped as to why it isn't working.

The relevant config section:

[stepper_z]
step_pin: P0.22
dir_pin: !P2.11
enable_pin: !P0.21
step_distance: .00125
endstop_pin: probe:z_virtual_endstop
position_max: 290

[probe]
pin: !P1.27
x_offset: 46
y_offset: 25.5
z_offset: 0
speed: 5

Attached the log file with a home and the led indicating before i hit the estop.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5781720/klippy.log)

Hywelmartin, 2021-01-07:
Use a pin without Pullup resistors on the board.. or remove
https://youtu.be/HOryknQnOd4?t=442

cyclonesurfer, 2021-01-08:
That actually explains it all.

cyclonesurfer, 2021-01-08:
I tried both the "servos" and "probe" pin, same problem.

If I don't ! the pin, its always triggered, if I do it doesn't recognize it.

langwadt, 2021-01-08:
which exact sensor is it? PNP, NPN?

cyclonesurfer, 2021-01-08:
PL-05N NPN NO

langwadt, 2021-01-09:
NPN needs a pullup

cyclonesurfer, 2021-01-09:
The xmin pin has a pull up on the board, doesn't work on that pin.

cyclonesurfer, 2021-01-13:
Please help, I doubt this is klippers fault, but I can't think of what I have hooked up wrong.

Hywelmartin, 2021-01-13:
have you tried the internal pullup??

`^`

cyclonesurfer, 2021-01-15:
lpc1768 has on board pull ups, not internal, but yes

cyclonesurfer, 2021-01-27:
I needed to install a signal diode on the signal wire, it works now.

If anyone else has a similar issue, on all NPN sensors you need a signal diode.

Please close

langwadt, 2021-01-27:
@cyclonesurfer a diode where? it makes no sense

closed 2021-02-26 by github-actions[bot]