Klipper3d/klipper #3738: CREALITY 3D V2.5.2 TMC2208 UART NO ISSUE

3Wings, 2021-01-07:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5783313/klippy.log)

Was just playing around, and wanted to share.....

[image]

[image]

[image]

[image]

[image]

georgebashi, 2021-01-07:
I can confirm, I did a similar trick and was able to find enough free pins to wire up all the UARTs of the TMC2208s.

3Wings, 2021-01-08:
Found some pins today, but need 1 more to get 4 x TM2208 and BL Touch up and running

[image]

3Wings, 2021-01-09:
Should be final version of UART solution for Creality 3D V2.5.2 
Will probably close issue in a few days if not mod does it sooner.
Save picture if you need it.

[image]

3Wings, 2021-01-09:
2021-01-09 16:18:26 | PWM_AUTO: 000e0024 PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14
-- | --
2021-01-09 16:18:26 | PWM_SCALE: 00000010 PWM_SCALE_SUM=16
2021-01-09 16:18:25 | PWMCONF: c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
2021-01-09 16:18:25 | DRV_STATUS: c00e0000 CS_ACTUAL=14 stealth=1 stst=1
2021-01-09 16:18:25 | CHOPCONF: 14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
2021-01-09 16:18:24 | MSCURACT: 00f7000c CUR_A=12 CUR_B=247
2021-01-09 16:18:24 | MSCNT: 00000008 MSCNT=8
2021-01-09 16:18:24 | TSTEP: 000fffff TSTEP=1048575
2021-01-09 16:18:24 | FACTORY_CONF: 0000000e FCLKTRIM=14
2021-01-09 16:18:23 | IOIN@TMC220x: 2000015d ENN=1 MS1=1 MS2=1 DIAG=1 PDN_UART=1 SEL_A=1(TMC220x) VERSION=0x20
2021-01-09 16:18:23 | OTP_READ: 0000000e OTP_FCLKTRIM=14
2021-01-09 16:18:23 | IFCNT: 0000000c IFCNT=12
2021-01-09 16:18:22 | GSTAT: 00000001 reset=1
2021-01-09 16:18:22 | GCONF: 000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
2021-01-09 16:18:22 | ========== Queried registers ==========
2021-01-09 16:18:21 | TPOWERDOWN: 00000014 TPOWERDOWN=20
2021-01-09 16:18:21 | TPWMTHRS: 0000003f TPWMTHRS=63
2021-01-09 16:18:21 | IHOLD_IRUN: 0008190e IHOLD=14 IRUN=25 IHOLDDELAY=8
2021-01-09 16:18:20 | ========== Write-only registers ==========
2021-01-09 16:18:20 | DUMP_TMC STEPPER=stepper_z
2021-01-09 16:18:15 | PWM_AUTO: 000e0024 PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14
2021-01-09 16:18:15 | PWM_SCALE: 0000000b PWM_SCALE_SUM=11
2021-01-09 16:18:15 | PWMCONF: c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
2021-01-09 16:18:14 | DRV_STATUS: c0090000 CS_ACTUAL=9 stealth=1 stst=1
2021-01-09 16:18:14 | CHOPCONF: 14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
2021-01-09 16:18:14 | MSCURACT: 00f7000c CUR_A=12 CUR_B=247
2021-01-09 16:18:14 | MSCNT: 00000008 MSCNT=8
2021-01-09 16:18:13 | TSTEP: 000fffff TSTEP=1048575
2021-01-09 16:18:13 | FACTORY_CONF: 0000000e FCLKTRIM=14
2021-01-09 16:18:13 | IOIN@TMC220x: 2000015d ENN=1 MS1=1 MS2=1 DIAG=1 PDN_UART=1 SEL_A=1(TMC220x) VERSION=0x20
2021-01-09 16:18:12 | OTP_READ: 0000000e OTP_FCLKTRIM=14
2021-01-09 16:18:12 | IFCNT: 00000006 IFCNT=6
2021-01-09 16:18:12 | GSTAT: 00000001 reset=1
2021-01-09 16:18:11 | GCONF: 000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
2021-01-09 16:18:11 | ========== Queried registers ==========
2021-01-09 16:18:11 | TPOWERDOWN: 00000014 TPOWERDOWN=20
2021-01-09 16:18:10 | TPWMTHRS: 00000026 TPWMTHRS=38
2021-01-09 16:18:10 | IHOLD_IRUN: 00081909 IHOLD=9 IRUN=25 IHOLDDELAY=8
2021-01-09 16:18:10 | ========== Write-only registers ==========
2021-01-09 16:18:09 | DUMP_TMC STEPPER=stepper_y
2021-01-09 16:18:05 | PWM_AUTO: 000e0024 PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14
2021-01-09 16:18:05 | PWM_SCALE: 0000000b PWM_SCALE_SUM=11
2021-01-09 16:18:05 | PWMCONF: c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
2021-01-09 16:18:04 | DRV_STATUS: c0090000 CS_ACTUAL=9 stealth=1 stst=1
2021-01-09 16:18:04 | CHOPCONF: 14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
2021-01-09 16:18:04 | MSCURACT: 00f7000c CUR_A=12 CUR_B=247
2021-01-09 16:18:03 | MSCNT: 00000008 MSCNT=8
2021-01-09 16:18:03 | TSTEP: 000fffff TSTEP=1048575
2021-01-09 16:18:03 | FACTORY_CONF: 00000011 FCLKTRIM=17
2021-01-09 16:18:02 | IOIN@TMC220x: 2000015d ENN=1 MS1=1 MS2=1 DIAG=1 PDN_UART=1 SEL_A=1(TMC220x) VERSION=0x20
2021-01-09 16:18:02 | OTP_READ: 00000011 OTP_FCLKTRIM=17
2021-01-09 16:18:02 | IFCNT: 00000006 IFCNT=6
2021-01-09 16:18:01 | GSTAT: 00000001 reset=1
2021-01-09 16:18:01 | GCONF: 000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
2021-01-09 16:18:01 | ========== Queried registers ==========
2021-01-09 16:18:00 | TPOWERDOWN: 00000014 TPOWERDOWN=20
2021-01-09 16:18:00 | TPWMTHRS: 00000026 TPWMTHRS=38
2021-01-09 16:18:00 | IHOLD_IRUN: 00081609 IHOLD=9 IRUN=22 IHOLDDELAY=8
2021-01-09 16:17:59 | ========== Write-only registers ==========
2021-01-09 16:17:59 | DUMP_TMC STEPPER=stepper_x

closed 2021-01-10 by 3Wings