Klipper3d/klipper #3742: Creality Ender 3 Pro Board v1.1.5

jorgeinglessis, 2021-01-08:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5788710/klippy.log)

Hi there,
I managed to upload the compiled firmware to the printer's board, but when get to the step of writing the  "status" command in the "confitguring klipper" section, I get the following:

```
Recv: // Unknown value 'PB12' in enumeration 'pin'
Recv: // This type of error is frequently caused by running an older
Recv: // version of the firmware on the micro-controller (fix by
Recv: // recompiling and flashing the firmware).
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Protocol error connecting to printer
Recv: // Klipper state: Not ready
Recv: !! Unknown value 'PB12' in enumeration 'pin'
```

Thank you very much for your attention.

Regards,

Jorge

Sineos, 2021-01-08:
Are you sure that you are using the right config for your board? For me this rather looks like https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-2018.cfg

jorgeinglessis, 2021-01-09:
Thank you very much for your quick response, 
I realized that although I have an ender 3 pro, I don't have the matching bard for the config I was trying. Then I tryied the config file you are suggesting, but y I'm getting a blank LCD screen. 
I have a creality V1.1.5 board with MC2208 stepper drivers, and I have seen many people  having issues with that borard, can you confirm that I can have fulll functionality with  my board/klipper firmware?, is ther any config file I can use a template with matches my board?

Thank you very much four your time,

Regards,

Jorge.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5790487/klippy.log)

Sineos, 2021-01-09:
Sorry, as I do not have the hardware nor do I know the type of display, I cannot assist any further.
Maybe https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#display-support can help.

jorgeinglessis, 2021-01-09:
Thank you for your answer, I''ll keep on searching.

Regards,
Jorge

crawler, 2021-01-17:
@jameswood I had the same issue when I used https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3pro-2020.cfg for my 1.1.4 board. But then I read the description comment, and it says that the config is for a 32-bit board. Ender 3 config (https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-2018.cfg) worked for my stock Ender 3 pro. The Maybe this will need some tuning, but at least this can be a point to start from.

jorgeinglessis, 2021-02-08:
I realized I was using the wrong config as a template. Thank you for your attention.

closed 2021-02-08 by jorgeinglessis