Klipper3d/klipper #3746: new to klipper

ankle-hunter, 2021-01-09:
hi everyone, i'm new to klipper, and i'm not having a great time. i've got a fresh install of octoprint 0.18 (because 0.17 isn't supported on the new raspberrypi 4b) and a skr mini e3 v1.2 mainboard. but i just cant get klipper to start on the printer. 
[image]

after doing 'make', i put the klipper.bin file on a sd card and changed the name to firmware.bin, and restarted the printer with the sd in it. i've been doing this for the last 2 days, trying different options in the 'klipper firmware configiration, but it just wont work.

Caewou, 2021-01-09:
If you haven't seen Chis Riley's video on klipper installation I recommend you to check it out. It seemed pretty clear to me and I make it worked through it. https://www.youtube.com/watch?v=i_541iD5Bj0&t=527s

ankle-hunter, 2021-01-09:
> If you haven't seen Chis Riley's video on klipper installation I recommend you to check it out. It seemed pretty clear to me and I make it worked through it. https://www.youtube.com/watch?v=i_541iD5Bj0&t=527s

yeah i followed every step, but no luck.

Sichikov, 2021-01-09:
If you had BL Touch, you should disconnect it and try without it.

ankle-hunter, 2021-01-10:
> If you had BL Touch, you should disconnect it and try without it.
> Original Ender pin out for BL-Touch is different from SKR E3 MINI and if connected halt flashing (You should change pins in connector)

Thanks! I'm gonna try that

ankle-hunter, 2021-01-11:
> If you had BL Touch, you should disconnect it and try without it.
> Original Ender pin out for BL-Touch is different from SKR E3 MINI and if connected halt flashing (You should change pins in connector)

Didn't work. Still no connection to the printer.

ankle-hunter, 2021-01-11:
> If you haven't seen Chis Riley's video on klipper installation I recommend you to check it out. It seemed pretty clear to me and I make it worked through it. https://www.youtube.com/watch?v=i_541iD5Bj0&t=527s

Did exactly this 3 times now, and it doesn't work

arturohernandez10, 2021-01-12:
You need to make sure you are using the right MCU port. Which raises the question, how are you connecting to the card. Best option I think is USB, and if that is the case, you may want to try the usb flashing. It gives you the MCU port you should use from the command line. It seems there was a recent change regarding the MCU config and it is tripping a few people. 

I just recently installed it and if the connection is lost I need to issue a reset before I see anything on the screen. They should have a lost connection or something else instead of a blank screen. The entire menu system I believe is controlled remotely so if you don't have a connection you don't have nothing.

Good Luck. This is really a good firmware, but it's rough around the edges.

closed 2021-01-16 by klipper-gitissuebot

ankle-hunter, 2021-01-16:
I forgot to change the usb target in the config...