Klipper3d/klipper #3748: [FR?] Menu item is not (automatically) refreshed?

hunagyp, 2021-01-10:
Hi Guys,
I do not know, if it's a bug or feature.
Let's take the following very basic [menu] snippet as example.
It creates a Pause menu entry if printer is not paused.
If I activate this menu item, it correctly calls the related macro. (y)
But, the Pause menu itself do not disappear and Resume item do not appear instead.
I thought, there is a "Refresh" in such a case, which would re-read the menu items and would update the LCD with current content.
If I go up to Main screen, then I go back 1 level down, Resume menu item is correctly listed instead of Pause.
That's why I suspect, there is no automatic refresh of menu items based on their "enable" field's value.
Is my expectation wrong in this case?
Is there maybe a {menu.refresh()} command like there is {menu.exit()} ?

```
[menu __main __pause]
type: command
enable: {not printer.pause_resume.is_paused}
name: Pause
index: 1
gcode:
    PAUSE

[menu __main __resume]
type: command
enable: {printer.pause_resume.is_paused}
name: Resume
index: 1
gcode:
    RESUME
```

An additional question: how would you create the following menu item with the available functions? Basically how one can create basic dialogs to interact with the user with simple questions and simple answers like Yes, No, Cancel in Jinja format?

```
[menu __main __misc __purge]
type: command
name: Purge
gcode:    
    PURGE
    <(full screen dialog)Ask the user, if purge was enough>
       <If user answers No, repeat PURGE macro>
       <if user answers Yes, finish actions here>
```
Thanks a lot in advance!

mcmatrix, 2021-01-11:
I think it's related to how menu updates itself.
List items are not updated when you are inside that list.
You need to leave from list folder or menu (if root list).

If the list items update were enabled then when new items were shown or hidden in real-time, the active item selection would jump and cause problems using the menu.

hunagyp, 2021-01-13:
@mcmatrix In my specific example, that would be the exact (final) goal, what you mentioned.
First, display Pause menu item. Then, when it is selected (activated), it should change itself (even in-place) to Resume menu item. But, for the moment, I just wanted to define 2 distinct menu items and show each of them them only based on is_paused property.
I feel, (the workaround) by exiting the active list and re-entering to force a refresh is not the best approach. Do not you agree?

closed 2021-01-17 by klipper-gitissuebot