Klipper3d/klipper #3752: Under voltage when connected directly to Ender 3v2 power supply

thelandinlook, 2021-01-10:
Good day lovelies,

Yesterday I did extensive voltage testing (even bricking one of my raspberry pi's) to power the Pi directly from the v2's PSU. I'm now getting this under voltage warning even having a consistent 5.1 volts through my most reliable buck converter (not flashing). I also believe it is receiving around 3.7amps. 
[image]
[image] (printing enclosure as soon as the printer is calibrated)

It doesn't seem to be affecting the printing performance in any way.
Attached to the pi I only have a small 5v fan and a pi camera. Additionally I've also covered the 5v pin on the USB to the printer.
Is this particular power supply not a good way to power the pi? Have I done something wrong?

[klippy.log](https://github.com/KevinOConnor/klipper/files/5792634/klippy.log)

Thanks xo

thelandinlook, 2021-01-10:
I thought I had fixed it, but no such luck.

JeffMedd, 2021-01-10:
Hi, I am also experiencing this issue with my SKR 1.4 with a Raspberry Pi 3 except this is causing printing issues. MCU communication problems where prints just stop dead. This has all started since upgrading to 0.9 of klipper where I had been printing without incident for months with the identical hardware setup under 0.8. I am just in the process of making a new sd card with 0.8 on it to test this theory.

mental405, 2021-01-28:
Unfortunately this sounds like an issue that is less Klipper and more Raspbian/Pi/Hardware. I am not really a big electronics guy, but I would venture to guess that it is possible that your pi is, in fact, experiencing a voltage drop due to _something_. Whether that be the ender power supply is unable to keep up (unlikely) or the buck converter is unable to keep up. I do know from my limited experience the that Raspberry Pi's and the Pi4 in particular are very particular about their power.

Sineos, 2021-01-28:
These DC-DC converter are mostly crap. From my experience, the PI-3B are starting to act up at around 4.8 and throws the UV warning at (IIRC) at 4.65V.
It depends on a lot of factors:
- Ripple of the DC-DC converter
- Voltage stability under load. Most of time the buffer caps are severely underrated --> Quick load changes on the PI are not properly compensated
- Connection of the PI (I see @thelandinlook has connected the power via GPIO. Good choice, just be careful you are circumventing all fuses / protections behind the USB port)
  - Worst invention ever is the flimsy Micro-USB port. You can expect some voltage drop here already
  - USB cable / quality. A cheap 50 cm AWG28 cable will cause a voltage drop of nearly 0.7 V at 2000 mA

A 1000µF electrolyte cap with a 10µF Tantal cap **in parallel** directly over your connection point to the PI should get you going.

Gussers1, 2021-02-06:
> 
> 
> Good day lovelies,
> 
> Yesterday I did extensive voltage testing (even bricking one of my raspberry pi's) to power the Pi directly from the v2's PSU. I'm now getting this under voltage warning (not flashing) even having a consistent 5.1 volts through my most reliable buck converter (I also get this with any voltage under 5.1). I also believe it is receiving around 3.7amps.
> <snip>
> Thanks xo

From the picture it looks like you have 1 USB going to the printer, but are supplying voltage to the pi directly, via the header pins? While you can do this, it's not recommended as there is NO protection. The USB/power port has protection built in. If you exceed the expected input voltage, it will simply not power on. No bricking. Via the header - you're on your own. 

I have a simple buck converter connected to my printer PS via two ring terminal connectors. They are connected to a pair of unused output terminals on the power supply. I then connected a female USB cable to the output from the buck converter and simply plug my pi (RPI 3B+) to this using a regular micro usb cable. I had to fiddle with the voltage to get it just right, but you can definitely get it to work reliably. I believe 5.1v was too high, so I settled on about 5.75v or so and do not get undervoltage issues. 

I'd suggest you use the power port intended to power the pi and create a corresponding socket to plug your usb cable in to from the buck converter. You could splice the other end of the cable directly in to the buck output, but I liked the idea of being able to disconnect everything without long cables being permanently attached.

closed 2021-03-07 by github-actions[bot]