Klipper3d/klipper #3754: Question - Should we config Octoprint to stay connected on firmware error?

ssotangkur, 2021-01-10:
I run into this catch-22 where to send the FIRMWARE_RESTART message from the terminal, I need to connect to the printer. But after an `M112` has been issued, attempts to connect to the printer result in an immediate disconnect.

I noticed that in the config: (wrench icon) --> Serial Connection --> Behavior we can change the default _What to do on firmware error (Error: or !!)_ option from `Disconnect from printer` to this:
[image]

I'm wondering if this is a good idea and if so could we include it in the Klipper docs to help the next person out.

Hywelmartin, 2021-01-10:
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper

ssotangkur, 2021-01-11:
Thanks, I must have just skipped over that line.

closed 2021-01-11 by ssotangkur